open-mmlab/OpenPCDet #1: Any plan for releasing the code of PV-RCNN?

memoiry, 2020-03-16:
Hi, Shaoshuai,

Thanks for your wonderful work! 

I'm wondering if there is a plan for releasing the code of PV-RCNN?

Thanks!

sshaoshuai, 2020-03-16:
PV-RCNN will be supported in the near future.

memoiry, 2020-03-16:
@sshaoshuai Is there an approximate time for the release? Thanks for your reply~

sshaoshuai, 2020-03-16:
maybe one or two months, I will find time to support it.

memoiry, 2020-03-16:
@sshaoshuai Great! Looking forward to your update.

Best

cslxiao, 2020-03-17:
Nice work!
The code is not compatible with the latest pytorch (1.4) due to the outdated spconv version.
Any plan to update the dependencies?

sshaoshuai, 2020-03-17:
Sorry, the codes are tested on pytorch 1.1 and pytorch 1.3, and currently I don't have much time to investigate the dependencies of spconv.

closed 2020-03-19 by sshaoshuai

willyd, 2020-03-25:
The fix to get spconv to work with PyTorch 1.4 is easy. See: https://github.com/traveller59/spconv/pull/105