open-mmlab/OpenPCDet #2: Any plan to support mode datasets such as nuScenes?

jinfagang, 2020-03-16:
Waymo and nuScenes are really good dataset to test.

sshaoshuai, 2020-03-16:
I will find time to support Waymo dataset recently, and contributions are welcomed to support nuScenes.

jinfagang, 2020-03-17:
that sounds very nice to support waymo.

Hoping for your workaround!

sharma-n, 2020-03-31:
Hey @sshaoshuai thanks for sharing your work! I would like to take a crack at training using the nuScenes dataset. I am planning to use the [nuscenes_to_kitti](https://github.com/lyft/nuscenes-devkit/blob/master/lyft_dataset_sdk/utils/export_kitti.py) code to rewrite the nuscenes dataset into the KITTI format. Just wanted your feedback on what are the things I will need to refactor. Here are some things I can see I will need to do:
1. Export the nuScenes dataset intothe KITTI format uisng [export_kitti.py](https://github.com/lyft/nuscenes-devkit/blob/master/lyft_dataset_sdk/utils/export_kitti.py)
2. Refactor `pcdet\datasets\kitti\kitti_dataset.py` with the correct filenames, etc.
3. Edit the `.yaml` config file to train new network.

I have a couple of quick questions:
1. Given that nuScenes detection classes are different from KITTI, will changing the `.yaml` file be sufficient to retrain, for example, PartA2?
2. The `ANCHOR_GENERATOR` in PartA2 is defined for each class separately. How did you decide what values work well for each class? For example, for Car we have `anchor_range: [0, -40, -1.78, 70.4, 40, -1.78]` while for Pedestrian we have `anchor_range: [0, -40, -0.6, 70.4, 40, -0.6]`. what do these values mean? Any way to calculate them for the different classes in the nuScenes dataset?

Cheers!

Nuri-benbarka, 2020-04-09:
Hi @jinfagang,
There is this repo [Waymo_Kitti_Adapter](https://github.com/Yao-Shao/Waymo_Kitti_Adapter) which converts the Waymo dataset into Kitti format. This might be a fast solution for you.
I hope you will find it useful.

rvalienter90, 2020-07-05:
Hi 
Also for another dataset, this repo https://github.com/yzhou377/argoverse-kitti-adapter converts the Argoverse dataset into KITTI format. With good results when using P-RCNN.  I hope you will find it useful.

sshaoshuai, 2020-07-30:
Hi all, the NuScenes dataset has been supported  with two strong baselines.

closed 2020-07-30 by sshaoshuai