open-mmlab/OpenPCDet #3: max batch per gpu of part_A2 is 2

buaaJeremyduan, 2020-03-16:
My gpu is nvidia-1080 Ti with 12 GB and the max lidar frames per gpu is 2 not 4.

sshaoshuai, 2020-03-16:
I have tested on 1080Ti with 11GB GPU memory and TITAN XP with 12GB GPU memory, both of them could support PartA2 with batch_size 4 for training. Have you used the default config?

buaaJeremyduan, 2020-03-17:
Yes, I used the default PartA2.yaml and install the version of spconv you mentioned in README.md   with pytorch 1.1. and CUDA 10.0.

buaaJeremyduan, 2020-03-17:
However, when I inference PartA2, I can set the batch_size to 4 and the gpu memory useage is about 7GB. Please leave the issue open to see if others have the same problem.

buaaJeremyduan, 2020-03-17:
When I train second with batch size 32(8 gpus): I got the result as follows, cls(car), iou=0.7:
bbox AP:94.9566, 89.9244, 88.9964
bev  AP:90.3231, 87.9408, 86.2429
3d   AP:88.7784, 78.5857, 77.2107
aos  AP:94.91, 89.78, 88.78

And When I train PartA2 with batch size 16(8 gpus): I got result as follows, 
bbox AP:90.7198, 89.3552, 89.0915
bev  AP:90.2864, 87.8155, 86.4079
3d   AP:89.6211, 79.1195, 78.5387
aos  AP:90.71, 89.25, 88.93
So, the batch size really makes a difference.

sshaoshuai, 2020-03-17:
Please double check that you install the `spconv 1.0` version (commit 8da6f96) I mentioned in README, since it consumes more GPU memory when training with the latest `spconv`.

buaaJeremyduan, 2020-03-18:
Sorry, My mistake. I reinstall the spconv and can train the PartA2 with 4 batch per gpu. But now the performance is :
bbox AP:96.5676, 89.3295, 89.0059
bev  AP:90.3915, 87.8028, 86.2632
3d   AP:89.5518, 79.2141, 78.6265
aos  AP:96.55, 89.24, 88.84.
which is better than ever but not as good as yours mentioned in READMD.md. I am still finetuning it.

closed 2020-03-25 by sshaoshuai