open-mmlab/OpenPCDet #5: Question about the  training strategy

jingyi-zhang, 2020-03-17:
Thanks for your wonderful work~
But I am puzzled on the training strategy, are the two stages of Part A^2 Net  trained separately like PointRCNN? 
I have never submitted results on kitti website, but I am trying to do a research job based on this dataset currently, so I wonder is it possible to train easy cases and hard cases  respectively?

sshaoshuai, 2020-03-17:
In this repo, the two-stage networks, like PartA2, are trained jointly, so you just need to run once for each model. 
It's possible to train easy and hard cases separately, but I think it may confuse the network since the objects of difficulty levels  have no clear split lines.

jingyi-zhang, 2020-03-17:
Thanks, I got it, you are too good!

closed 2020-03-17 by jingyi-zhang