open-mmlab/OpenPCDet #9: something wrong when running train.py

Cjmmmmmm, 2020-03-21:
File "/home/PCDet/pcdet/datasets/kitti/kitti_dataset.py", line 398, in __getitem__
    sample_idx = info['point_cloud']['lidar_idx']
KeyError: 'lidar_idx'

closed 2020-03-22 by Cjmmmmmm

PolyU-IPNL, 2020-05-22:
I have encounter the same problem, how would you solve it?

wind19970924, 2020-10-05:
I have encounter the same problem, how would you solve it?

Rajat-Mehta, 2020-11-22:
@Cjmmmmmm @wind19970924 @PolyU-IPNL did any one of you solve this issue? If yes, how? I am also getting this error when training the pvrcnn model on kitti dataset.

twangnh, 2021-01-09:
same problem

Dinghow, 2021-05-08:
same problem

dailingjun, 2021-11-18:
The pkl files may be generated by mmdetection3d in which the keys are different. 
Delete or move all the pkl files. Re-generate the data infos by running the following command:
python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml