open-mmlab/OpenPCDet #10: RuntimeError: copy_if failed to synchronize: device-side assert triggered

Shaoqing26, 2020-03-22:
2020-03-22 16:34:58,582   INFO  **********************Start logging**********************
2020-03-22 16:34:58,584   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-03-22 16:34:58,584   INFO  cfg_file         cfgs/pointpillar.yaml
2020-03-22 16:34:58,585   INFO  data_dir         None
2020-03-22 16:34:58,586   INFO  batch_size       4
2020-03-22 16:34:58,586   INFO  epochs           80
2020-03-22 16:34:58,587   INFO  workers          4
2020-03-22 16:34:58,588   INFO  extra_tag        default
2020-03-22 16:34:58,588   INFO  ckpt             pointpillar.pth
2020-03-22 16:34:58,589   INFO  mgpus            False
2020-03-22 16:34:58,589   INFO  launcher         none
2020-03-22 16:34:58,590   INFO  tcp_port         18888
2020-03-22 16:34:58,591   INFO  local_rank       0
2020-03-22 16:34:58,592   INFO  set_cfgs         None
2020-03-22 16:34:58,592   INFO  max_waiting_mins 30
2020-03-22 16:34:58,593   INFO  start_epoch      0
2020-03-22 16:34:58,593   INFO  eval_tag         default
2020-03-22 16:34:58,594   INFO  eval_all         False
2020-03-22 16:34:58,595   INFO  ckpt_dir         None
2020-03-22 16:34:58,596   INFO  save_to_file     False
2020-03-22 16:34:58,596   INFO  cfg.ROOT_DIR: /media/buaa/My Passport/PCDet
2020-03-22 16:34:58,597   INFO  cfg.LOCAL_RANK: 0
2020-03-22 16:34:58,598   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-03-22 16:34:58,599   INFO  
cfg.DATA_CONFIG = edict()
2020-03-22 16:34:58,600   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-03-22 16:34:58,601   INFO  cfg.DATA_CONFIG.DATA_DIR: data/kitti
2020-03-22 16:34:58,602   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-03-22 16:34:58,602   INFO  
cfg.DATA_CONFIG.NUM_POINT_FEATURES = edict()
2020-03-22 16:34:58,603   INFO  cfg.DATA_CONFIG.NUM_POINT_FEATURES.total: 4
2020-03-22 16:34:58,604   INFO  cfg.DATA_CONFIG.NUM_POINT_FEATURES.use: 4
2020-03-22 16:34:58,605   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
2020-03-22 16:34:58,605   INFO  cfg.DATA_CONFIG.MASK_POINTS_BY_RANGE: True
2020-03-22 16:34:58,606   INFO  
cfg.DATA_CONFIG.TRAIN = edict()
2020-03-22 16:34:58,606   INFO  cfg.DATA_CONFIG.TRAIN.INFO_PATH: ['data/kitti/kitti_infos_train.pkl']
2020-03-22 16:34:58,607   INFO  cfg.DATA_CONFIG.TRAIN.SHUFFLE_POINTS: True
2020-03-22 16:34:58,608   INFO  cfg.DATA_CONFIG.TRAIN.MAX_NUMBER_OF_VOXELS: 16000
2020-03-22 16:34:58,608   INFO  
cfg.DATA_CONFIG.TEST = edict()
2020-03-22 16:34:58,609   INFO  cfg.DATA_CONFIG.TEST.INFO_PATH: ['data/kitti/kitti_infos_val.pkl']
2020-03-22 16:34:58,610   INFO  cfg.DATA_CONFIG.TEST.SHUFFLE_POINTS: False
2020-03-22 16:34:58,610   INFO  cfg.DATA_CONFIG.TEST.MAX_NUMBER_OF_VOXELS: 40000
2020-03-22 16:34:58,611   INFO  
cfg.DATA_CONFIG.AUGMENTATION = edict()
2020-03-22 16:34:58,612   INFO  
cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT = edict()
2020-03-22 16:34:58,613   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT.ENABLED: True
2020-03-22 16:34:58,613   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT.GT_LOC_NOISE_STD: [1.0, 1.0, 0.1]
2020-03-22 16:34:58,614   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT.GT_ROT_UNIFORM_NOISE: [-0.78539816, 0.78539816]
2020-03-22 16:34:58,615   INFO  
cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE = edict()
2020-03-22 16:34:58,615   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE.ENABLED: True
2020-03-22 16:34:58,616   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE.GLOBAL_ROT_UNIFORM_NOISE: [-0.78539816, 0.78539816]
2020-03-22 16:34:58,616   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE.GLOBAL_SCALING_UNIFORM_NOISE: [0.95, 1.05]
2020-03-22 16:34:58,617   INFO  
cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER = edict()
2020-03-22 16:34:58,618   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.ENABLED: True
2020-03-22 16:34:58,619   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.DB_INFO_PATH: ['data/kitti/kitti_dbinfos_train.pkl']
2020-03-22 16:34:58,620   INFO  
cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.PREPARE = edict()
2020-03-22 16:34:58,621   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.PREPARE.filter_by_difficulty: [-1]
2020-03-22 16:34:58,621   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.PREPARE.filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5']
2020-03-22 16:34:58,622   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.RATE: 1.0
2020-03-22 16:34:58,623   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.SAMPLE_GROUPS: ['Car:15', 'Pedestrian:10', 'Cyclist:10']
2020-03-22 16:34:58,624   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.USE_ROAD_PLANE: True
2020-03-22 16:34:58,624   INFO  
cfg.DATA_CONFIG.VOXEL_GENERATOR = edict()
2020-03-22 16:34:58,625   INFO  cfg.DATA_CONFIG.VOXEL_GENERATOR.MAX_POINTS_PER_VOXEL: 32
[... 112 lines cut ...]
    main()
  File "test.py", line 177, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id)
  File "test.py", line 59, in eval_single_ckpt
    model, test_loader, epoch_id, logger, result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/media/buaa/My Passport/PCDet/tools/eval_utils/eval_utils.py", line 46, in eval_one_epoch
    pred_dicts, ret_dict = model(input_dict)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/buaa/My Passport/PCDet/pcdet/models/detectors/pointpillar.py", line 34, in forward
    rpn_ret_dict = self.forward_rpn(**input_dict)
  File "/media/buaa/My Passport/PCDet/pcdet/models/detectors/pointpillar.py", line 18, in forward_rpn
    output_shape=self.grid_size[::-1]
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/buaa/My Passport/PCDet/pcdet/models/rpn/pillar_scatter.py", line 32, in forward
    this_coords = coords[batch_mask, :]
RuntimeError: copy_if failed to synchronize: device-side assert triggered
eval:   0%|

when i run
python3 test.py --cfg_file cfgs/pointpillar.yaml --batch_size 4 --ckpt pointpillar.pth

I met the problem above ,How can i solved the problem?
THK

Shaoqing26, 2020-03-22:
### **ADN,when i set batch_size = 1,there has another problem as:**

2020-03-22 19:05:12,088   INFO  **********************Start logging**********************
2020-03-22 19:05:12,089   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-03-22 19:05:12,090   INFO  cfg_file         cfgs/pointpillar.yaml
2020-03-22 19:05:12,091   INFO  data_dir         None
2020-03-22 19:05:12,092   INFO  batch_size       1
2020-03-22 19:05:12,092   INFO  epochs           80
2020-03-22 19:05:12,093   INFO  workers          4
2020-03-22 19:05:12,094   INFO  extra_tag        default
2020-03-22 19:05:12,094   INFO  ckpt             pointpillar.pth
2020-03-22 19:05:12,095   INFO  mgpus            False
2020-03-22 19:05:12,095   INFO  launcher         none
2020-03-22 19:05:12,096   INFO  tcp_port         18888
2020-03-22 19:05:12,096   INFO  local_rank       0
2020-03-22 19:05:12,097   INFO  set_cfgs         None
2020-03-22 19:05:12,098   INFO  max_waiting_mins 30
2020-03-22 19:05:12,098   INFO  start_epoch      0
2020-03-22 19:05:12,099   INFO  eval_tag         default
2020-03-22 19:05:12,100   INFO  eval_all         False
2020-03-22 19:05:12,100   INFO  ckpt_dir         None
2020-03-22 19:05:12,101   INFO  save_to_file     False
2020-03-22 19:05:12,101   INFO  cfg.ROOT_DIR: /media/buaa/My Passport/PCDet
2020-03-22 19:05:12,102   INFO  cfg.LOCAL_RANK: 0
2020-03-22 19:05:12,103   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-03-22 19:05:12,104   INFO  
cfg.DATA_CONFIG = edict()
2020-03-22 19:05:12,104   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-03-22 19:05:12,105   INFO  cfg.DATA_CONFIG.DATA_DIR: data/kitti
2020-03-22 19:05:12,106   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-03-22 19:05:12,106   INFO  
cfg.DATA_CONFIG.NUM_POINT_FEATURES = edict()
2020-03-22 19:05:12,107   INFO  cfg.DATA_CONFIG.NUM_POINT_FEATURES.total: 4
2020-03-22 19:05:12,107   INFO  cfg.DATA_CONFIG.NUM_POINT_FEATURES.use: 4
2020-03-22 19:05:12,108   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
2020-03-22 19:05:12,109   INFO  cfg.DATA_CONFIG.MASK_POINTS_BY_RANGE: True
2020-03-22 19:05:12,109   INFO  
cfg.DATA_CONFIG.TRAIN = edict()
2020-03-22 19:05:12,110   INFO  cfg.DATA_CONFIG.TRAIN.INFO_PATH: ['data/kitti/kitti_infos_train.pkl']
2020-03-22 19:05:12,111   INFO  cfg.DATA_CONFIG.TRAIN.SHUFFLE_POINTS: True
2020-03-22 19:05:12,111   INFO  cfg.DATA_CONFIG.TRAIN.MAX_NUMBER_OF_VOXELS: 16000
2020-03-22 19:05:12,112   INFO  
cfg.DATA_CONFIG.TEST = edict()
2020-03-22 19:05:12,113   INFO  cfg.DATA_CONFIG.TEST.INFO_PATH: ['data/kitti/kitti_infos_val.pkl']
2020-03-22 19:05:12,113   INFO  cfg.DATA_CONFIG.TEST.SHUFFLE_POINTS: False
2020-03-22 19:05:12,114   INFO  cfg.DATA_CONFIG.TEST.MAX_NUMBER_OF_VOXELS: 40000
2020-03-22 19:05:12,115   INFO  
cfg.DATA_CONFIG.AUGMENTATION = edict()
2020-03-22 19:05:12,115   INFO  
cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT = edict()
2020-03-22 19:05:12,116   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT.ENABLED: True
2020-03-22 19:05:12,116   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT.GT_LOC_NOISE_STD: [1.0, 1.0, 0.1]
2020-03-22 19:05:12,117   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_PER_OBJECT.GT_ROT_UNIFORM_NOISE: [-0.78539816, 0.78539816]
2020-03-22 19:05:12,118   INFO  
cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE = edict()
2020-03-22 19:05:12,118   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE.ENABLED: True
2020-03-22 19:05:12,119   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE.GLOBAL_ROT_UNIFORM_NOISE: [-0.78539816, 0.78539816]
2020-03-22 19:05:12,120   INFO  cfg.DATA_CONFIG.AUGMENTATION.NOISE_GLOBAL_SCENE.GLOBAL_SCALING_UNIFORM_NOISE: [0.95, 1.05]
2020-03-22 19:05:12,120   INFO  
cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER = edict()
2020-03-22 19:05:12,121   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.ENABLED: True
2020-03-22 19:05:12,122   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.DB_INFO_PATH: ['data/kitti/kitti_dbinfos_train.pkl']
2020-03-22 19:05:12,122   INFO  
cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.PREPARE = edict()
2020-03-22 19:05:12,123   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.PREPARE.filter_by_difficulty: [-1]
2020-03-22 19:05:12,124   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.PREPARE.filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5']
2020-03-22 19:05:12,125   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.RATE: 1.0
2020-03-22 19:05:12,126   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.SAMPLE_GROUPS: ['Car:15', 'Pedestrian:10', 'Cyclist:10']
2020-03-22 19:05:12,127   INFO  cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.USE_ROAD_PLANE: True
2020-03-22 19:05:12,128   INFO  
[... 108 lines cut ...]
    model, test_loader, epoch_id, logger, result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/media/buaa/My Passport/PCDet/tools/eval_utils/eval_utils.py", line 46, in eval_one_epoch
    pred_dicts, ret_dict = model(input_dict)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/buaa/My Passport/PCDet/pcdet/models/detectors/pointpillar.py", line 34, in forward
    rpn_ret_dict = self.forward_rpn(**input_dict)
  File "/media/buaa/My Passport/PCDet/pcdet/models/detectors/pointpillar.py", line 22, in forward_rpn
    **{'gt_boxes': kwargs.get('gt_boxes', None)}
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
[... cut ...]
    result = self.forward(*input, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/conv.py", line 345, in forward
    return self.conv2d_forward(input, self.weight)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/conv.py", line 342, in conv2d_forward
    self.padding, self.dilation, self.groups)
RuntimeError: cuDNN error: CUDNN_STATUS_MAPPING_ERROR
eval:   0%|                                                                                      | 4/3769 [00:05<1:28:06,  1.40s/it]

chreisinger, 2020-03-24:
I had the same problem and I figured out that this error comes from _pcdet/model/rpn/pillar_scatter.py_.

The error occurs in line 32 and 33:
[32] `indices = this_coords[:, 1] * nz + this_coords[:, 2] * nx + this_coords[:, 3]`
[33] `indices = indices.type(torch.long)`

This casting produces **negative indices** which leads to an error. My fix looks like this: 
indices = this_coords[:, 1].type(torch.long) * nz + this_coords[:, 2].type(torch.long) * nx + this_coords[:, 3].type(torch.long)

However, I don't understand why 'this_coords.type(torch.long)' before the mentioned lines doesn't fix the error.

Shaoqing26, 2020-03-25:
> I had the same problem and I figured out that this error comes from _pcdet/model/rpn/pillar_scatter.py_.
>
> The error occurs in line 32 and 33:
> [32] `indices = this_coords[:, 1] * nz + this_coords[:, [...]

### **It worked for me ，I am very appreciate your advice, but I also wana to know why it worked,Can i explain it for a while?**

lan-sky, 2020-05-23:
Do you get the good result？
I fix like this：
#indices = this_coords[:, 1] * nz + this_coords[:, 2] * nx + this_coords[:, 3]
#indices = indices.type(torch.long)
indices = this_coords[:, 1].type(torch.long) * nz + this_coords[:, 2].type(torch.long) * nx + this_coords[:, 3].type(torch.long)
indices = indices.type(torch.long)
But i get the wrong result：
2020-05-23 01:57:55,972   INFO  cfg.MODEL.TEST.BOX_FILTER.USE_IMAGE_AREA_FILTER: True
2020-05-23 01:57:55,972   INFO  cfg.MODEL.TEST.BOX_FILTER.LIMIT_RANGE: [0, -40, -3.0, 70.4, 40, 3.0]
2020-05-23 01:57:55,972   INFO  cfg.TAG: pointpillar
2020-05-23 01:57:56,044   INFO  Loading KITTI dataset
2020-05-23 01:57:56,331   INFO  Total samples for KITTI dataset: 3769
2020-05-23 01:58:02,530   INFO  ==> Loading parameters from checkpoint ./weight/pointpillar.pth to GPU
2020-05-23 01:58:03,157   INFO  ==> Done (loaded 127/127)
2020-05-23 01:58:03,182   INFO  *************** EPOCH no_number EVALUATION *****************
2020-05-23 02:02:46,837   INFO  *************** Performance of EPOCH no_number *****************
2020-05-23 02:02:46,842   INFO  Generate label finished(sec_per_example: 0.0752 second).
2020-05-23 02:02:46,844   INFO  Average predicted number of objects(3769 samples): 25.871
2020-05-23 02:03:11,634   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
@Shaoqing26

Shaoqing26, 2020-05-23:
> Do you get the good result？
> I fix like this：
> #indices = this_coords[:, 1] * nz + this_coords[:, 2] * nx + this_coords[:, 3]
> #indices = indices.type(torch.long)
> indices = this_coords[:, [...]

I got the result same as author，maybe  you used  wrong model path，I met the same problem as you，And the reason is my model path is wrong。

lan-sky, 2020-05-23:
> I got the result same as author，maybe you used wrong model path，I met the same problem as you，And the reason is my model path is wrong。

Can you show me the model path and the command？ If it's the wrong path，there will be a FileNotFoundErro。Thanks😭

lan-sky, 2020-05-23:
This is my model path
[image]
and i run ：
python3 test.py --cfg_file cfgs/pointpillar.yaml --batch_size 4 --ckpt pointpillar.pth 
Whats the matter？I have spent plenty of time on looking at code but I haven't figure out the issue.
Thanks @Shaoqing26

Crowbar97, 2020-05-25:
> This is my model path
> [image]
> and i run ：
> python3 test.py --cfg_file cfgs/pointpillar.yaml --batch_size 4 --ckpt pointpillar.pth
> Whats the matter？I have spent plenty of time on looking at code but I haven't figure out the issue.
> Thanks @Shaoqing26

Thanks to @chreisinger, his suggested code fix works for me as well

@lan-sky, are you sure that you correctly fix the code?
The file `pcdet/models/rpn/pillar_scatter.py` correction (_line 33_) should be like this:
```python
# indices = this_coords[:, 1] * nz + this_coords[:, 2] * nx + this_coords[:, 3]
indices = this_coords[:, 1].type(torch.long) * nz + this_coords[:, 2].type(torch.long) * nx + this_coords[:, 3].type(torch.long)
```
By the way, have you tried to launch another model, e.g. `Part-A2`?

lan-sky, 2020-05-26:
> Thanks to @chreisinger, his suggested code fix works for me as well
>
> @lan-sky, are you sure that you correctly fix the code?
> The file `pcdet/models/rpn/pillar_scatter.py` correction (_line 33_) [...]

Yes, I think I fix the code same with you:
[image]
With this modification, the program can be tested, but I cant get the correct result
@Crowbar97  How can i solve this problem, THX

Crowbar97, 2020-05-26:
> Yes, I think I fix the code same with you:
> [image]
> With this modification, the program can be tested, but I cant get the correct result
> @Crowbar97 How can i solve this problem, THX

Zero results can be due to next issues:
- Wrong / corrupted config `.yaml` file
- Wrong `test.py` script arguments
- Wrong `.pth` model
- Wrong installed dependencies (make sure you install the `spconv v1.0` (`commit 8da6f96`) instead of the latest one)
- Corrupted dataset (or may be some mess in files and directories)
- ...

@lan-sky, so I think it's good idea to reinstall this repo from scratch and try once again

lan-sky, 2020-05-26:
> Zero results can be due to next issues:
>
> * Wrong / corrupted config `.yaml` file
> * Wrong `test.py` script arguments
> * Wrong `.pth` model
> * Wrong installed dependencies (make sure you install the [...]

Ok，I will try to reinstall. BTY, I used the spconv v1.1 Because I saw author was update to spconv 1.1
Thank you for you advice @Crowbar97

closed 2020-07-21 by sshaoshuai