open-mmlab/OpenPCDet #12: AttributeError: int object has no attribute 'shape'

Gang-Chen-China, 2020-03-23:
when is run second , it shows above error at pcdet/models/detectors/detector3d.py/line 303

```
for key, val in model_state_disk.items():
        if key in self.state_dict() and self.state_dict()[key].shape == model_state_disk[key].shape:
        update_model_state[key] = val
        # logger.info('Update weight %s: %s' % (key, str(val.shape)))
```

do you know how to solve it?

sshaoshuai, 2020-03-25:
Which model does you load? Could you please print out the failed key?

sshaoshuai, 2020-03-25:
I have uploaded a new version of `1second.pth` to fix this problem, please download the new one.

Gang-Chen-China, 2020-03-25:
> 
> 
> I have uploaded a new version of `second.pth` to fix this problem, please download the new one.
thanks , i will try later!

Gang-Chen-China, 2020-03-25:
> 
> 
> I have uploaded a new version of `second.pth` to fix this problem, please download the new one.

thanks , i will try later.

closed 2020-03-25 by sshaoshuai