open-mmlab/OpenPCDet #14: Semanttic output

abhigoku10, 2020-03-24:
@sshaoshuai hello thanks for open sourcing the code i have few queries 
Q1 in the paper you had mentioned about the inference time as 14fps on what sys
Q2 can you obtain the intermediate semantic segmentation output of point clouds ? i show which part of the code does it 
Q3 Can we train your architecture onreduced point clouds will the performance varry 
Q4 Can we train of static / immovable objects 

Thanks in advance

sshaoshuai, 2020-03-25:
Q1: Inference time is tested on Tesla V100 GPU card.
Q2: The semantic segmentation codes are https://github.com/sshaoshuai/PCDet/blob/master/pcdet/models/rpn/rpn_unet.py#L500
Q3: Yes, you could train on the reduced point clouds for KITTI datasets by removing the `get_fov_flag` codes https://github.com/sshaoshuai/PCDet/blob/master/pcdet/datasets/kitti/kitti_dataset.py#L414, the performance should be similar.
Q4: What does you mean for static/immovable objects?

abhigoku10, 2020-03-25:
@sshaoshuai Q4 static objects mean trees building poles

sshaoshuai, 2020-03-25:
If you have GT labels for these objects, you could try to train with them for detecting them. But I think these objects generally are not  represented by 3D bounding boxes, and segmentation is more naturally.

closed 2020-03-31 by sshaoshuai