open-mmlab/OpenPCDet #16: How to reproduce reported performance

willyd, 2020-03-25:
Thanks for this nice code base.

Can you provided instructions on how to reproduce the reported performance on the `val` set. 

I tried:
python test.py --cfg_file ${CONFIG_FILE} --batch_size 4 --ckpt ${CKPT}

with the pretrained model and I get:
```
Car_3d_easy = 54.303261947978974
Car_3d_moderate = 38.349761507380684
Car_3d_hard = 36.39773833709999
```
with a model trained with the default config:
```
Car_3d_easy = 76.20793977206878                                                
Car_3d_moderate = 60.98983744758463                                                          
Car_3d_hard = 57.58892404665682 
```
which is far from the reported values. Am I doing something wrong? Is the provided config the one with which you report the performance?

Thanks

sshaoshuai, 2020-03-25:
The provided pretrained models could achieve exactly the same reported performance. Have you modified the codes? I think you should make true to achieve the same reported performance with the given model firstly.

chreisinger, 2020-03-25:
Hi, I was able to reproduce similar results as reported by training the network on my own (SECOND, Pointpillars) with the proposed commands.

willyd, 2020-03-25:
Thanks for the quick reply. I did not modify the code except for a small change in spconv to make it work with pytorch 1.4. I only tried the default config for pointpillars, but did not get the error mentionned by @chreisinger. I guess I could try with 1.1 or 1.3. 

@sshaoshuai @chreisinger What pytorch, numpy versions are you using?

willyd, 2020-03-25:
Forgot to mention I only have 4 gpus so I changed the batch size to 28 per GPU instead of 32. I guess should have used 16 as recommended instead. 

But that would not explain the poor results obtained with the pretained model.

chreisinger, 2020-03-25:
Okay, I use Pytorch 1.3, the newest version of numpy and an unchanged version of spconv. By the way, I use a single 1080Ti.

willyd, 2020-03-25:
The issue was with a corrupted KITTI dataset I got from someone else... Sorry for the noise and thanks for the support. I can reproduce the exact same numbers with the PointPillars pretrained model with pytorch 1.4.

closed 2020-03-25 by willyd