open-mmlab/OpenPCDet #18: Dockerfile

trn84, 2020-03-26:
Hello,

thank you for the great work. I think it is great that you are integrating many different models and networks into this framework. I am working on an automated build for your project since I had some issues with the installation. Maybe you want to integrate the Dockerfile at: 

https://github.com/trn84/pcdet-docker

Into this projects. It is basically taken and modified from the second.pytorch repo. I have tested the evaluation and some training but it is still in prelim stage.

BR

sshaoshuai, 2020-03-27:
Hi BR, 
Thank you for building the docker for PCDet, and I think it provides an alternative convenient way for using this repo. You could send a pull request when the docker is ready or I could add some comments on the README to refer to your repo.

trn84, 2020-03-28:
Hi,

okay I have created a pull request...Guess it is best to place the Dockerfile here :-)

BR

sshaoshuai, 2020-03-29:
Hi BR, 

I think it should be the best place and I have merged it into the master branch. 
Thank you for the contribution!

Shaoshuai

closed 2020-03-29 by sshaoshuai