open-mmlab/OpenPCDet #21: I have an error that I cant solve

MohamedAboushnief, 2020-03-29:
when I run the train command or the test command I get this error 

**File "/home/mohamed/Mohamed/HTW/lib/python3.6/site-packages/spconv/utils/__init__.py", line 65, in points_to_voxel
    assert block_filtering is False
AssertionError**

and when I change the flag of the assert in the code in __init__.py to see what will happen I get this error

**File "/home/mohamed/Mohamed/HTW/lib/python3.6/site-packages/spconv/utils/__init__.py", line 94, in points_to_voxel
    coors_range.tolist(), max_points, max_voxels)
ValueError: array has incorrect number of dimensions: 2; expected 3**

please help,  I dont know what is wrong

muzi2045, 2020-03-30:
using spconv 1.0 not spconv 1.1

MohamedAboushnief, 2020-03-30:
Every time I try to install the spconv 1.0 library , the 1.1 version gets installed. I do what the documentation says and then I find in the ./dist  this file (spconv-1.1-cp36-cp36m-linux_x86_64.whl) which is for installing the 1.1 version. I don't know why this happens for me. I need the spconv 1.0 version but I can't get it.

Nuri-benbarka, 2020-03-30:
In the spcnv git repo you cloned, use this command which will let the git repo go to version 1:

```
git checkout 8da6f96
```

Then continue with the same installation instructions you did.

closed 2020-04-24 by sshaoshuai

LittleYuanzi, 2020-06-25:
new version has this question using spconv1.1.
`2020-06-25 09:14:28,205   INFO  **********************Start training cfgs/kitti_models/pv_rcnn(ttt)**********************
epochs:   0%|                                            | 0/16 [00:00<?, ?it/s]
train:   0%|                                           | 0/3712 [00:00<?, ?it/s]
epochs:   0%|                                            | 0/16 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/tools/train.py", line 166, in <module>
    main()
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/tools/train.py", line 158, in main
    merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/tools/train_utils/train_utils.py", line 92, in train_model
    dataloader_iter=dataloader_iter
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/tools/train_utils/train_utils.py", line 18, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 819, in __next__
    return self._process_data(data)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 846, in _process_data
    data.reraise()
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/_utils.py", line 369, in reraise
    raise self.exc_type(msg)
AssertionError: Caught AssertionError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/pcdet/datasets/kitti/kitti_dataset.py", line 388, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/pcdet/datasets/dataset.py", line 94, in prepare_data
    data_dict=data_dict
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/pcdet/datasets/processor/data_processor.py", line 87, in forward
    data_dict = cur_processor(data_dict=data_dict)
  File "/home/jty/pcdet/pvrcnn/OpenLidarPerceptron/pcdet/datasets/processor/data_processor.py", line 65, in transform_points_to_voxels
    voxels, coordinates, num_points = voxel_generator.generate(points)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/spconv/utils/__init__.py", line 164, in generate
    or self._max_voxels, self._full_mean)
  File "/home/jty/anaconda3/envs/pointpillars/lib/python3.7/site-packages/spconv/utils/__init__.py", line 65, in points_to_voxel
    assert block_filtering is False`

LittleYuanzi, 2020-06-25:
> new version has this question using spconv1.1.
> `2020-06-25 09:14:28,205 INFO **********************Start training cfgs/kitti_models/pv_rcnn(ttt)********************** epochs: 0%| | 0/16 [00:00<?, [...]

while spconv1.0 is ok