open-mmlab/OpenPCDet #24: rcnn_loss don't decline when training on my custum dataset

xzswater, 2020-04-01:
As shown on the fig, **### rcnn_loss don't decline** sharply; on the contrary, rpn_loss worked well.
So, what are the possible reasons ? Thank you for your suggestion.

[image]

sshaoshuai, 2020-04-02:
How about the final mAP?  Sometimes the rcnn loss curve could not well indicate the training process of the network. Also, the rcnn_fg_num seems too low. Maybe you should check the quality of RPN proposals. And also you could modify the RCNN.TARGET_CONFIG.[REG_FG_THRESH, CLS_FG_THRESH, CLS_BG_THRESH] according to your dataset or classes.

xzswater, 2020-04-08:
@sshaoshuai  Thank you for your valuable advice, i did found some bugs and fixed them !
And i modified the PartA2.yaml configs to fit my custum dataset.
[image]

### But, the recall_rcnn_* and recall_roi_* still very low when i eval on test sub-dataset. Could you help to provide some more suggestions ? Looking forward to your reply .

[image]

In addition, our pointclouds is labeld all 360 degrees, without 2d images. So, i set POINT_CLOUD_RANGE to  [-60, -60, -1.5, 60, 60, 2.5].
Because no 2d images, i'm modifying kitti_object_eval_python/eval.py, so can't provide the mAPs now .

The tensorboard graphs :
[image]

closed 2020-05-20 by xzswater