open-mmlab/OpenPCDet #25: Error kernelSize.size() == coorDim assert failed

sharma-n, 2020-04-02:
Hi,
I am trying to use the pre-trained PartA^2_fc model to run it on the NuScenes dataset. I have successfully generated the required annotation files, such as `infos_train.pkl`, `infos_test.pkl` and so on.

However, when I try to run the code by using the `PartA2_fc.yaml`, I get the following error:
```
File ".../lib/python3.7/site-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File "PCDet/pcdet/models/detectors/PartA2_net.py", line 106, in forward
    rpn_ret_dict = self.forward_rpn(**input_dict)
  File "PCDet/pcdet/models/detectors/PartA2_net.py", line 33, in forward_rpn
    **kwargs
  File ".../lib/python3.7/site-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File "PCDet/pcdet/models/rpn/rpn_unet.py", line 471, in forward
    x = self.conv_input(input_sp_tensor)
  File ".../lib/python3.7/site-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File ".../lib/python3.7/site-packages/spconv/modules.py", line 123, in forward
    input = module(input)
  File ".../lib/python3.7/site-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File ".../lib/python3.7/site-packages/spconv/conv.py", line 151, in forward
    self.stride, self.padding, self.dilation, self.output_padding, self.subm, self.transposed, grid=input.grid)
  File ".../lib/python3.7/site-packages/spconv/ops.py", line 89, in get_indice_pairs
    stride, padding, dilation, out_padding, int(subm), int(transpose))
RuntimeError: PCDet/spconv/include/spconv/spconv_ops.h 40
kernelSize.size() == coorDim assert faild. error
```
If I am guessing correctly, this is because after voxelization of the point cloud, the NN model expects a particular dimension of voxel grid. I can probably correct this by changing the `VOXEL_SIZE` in `PartA2_fc.yaml`. Is this true? How should I calculate how to set this parameter?

Thanks for your help!

sshaoshuai, 2020-04-02:
I don't know which layer it is, and how about the shape of the input tensor to this layer?  
Also, the given model use (Nx4) as the input with xyz+ one intensity. You should make sure your input is correct. 
The intensity and point density of KITTI and NuScenes may be different so you could not directly achieve a well results on the NuScene dataset with the given KITTI model.

sharma-n, 2020-04-06:
Hey, thanks for the hint. Yes, you were right, I was inferring on 1 point cloud at a time, and the coordinates (and points) were missing the extra dimension added to keep track of different point clouds when batch processing.

closed 2020-04-06 by sharma-n