open-mmlab/OpenPCDet #28: Is anyone evaluate successfully on your custom dataset?

KevinDu1, 2020-04-03:
I test my own dataset using the following code 
`python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2.pth \ 
    --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True`
But the results is bad that the detected results don't in the correct situation whether in the pointcloud or in the images. Does anyone know where this problem comes from and how I can solve it? 
 
[image]

sshaoshuai, 2020-04-06:
What's the coordinate system of your input point cloud? It should be the same with the provided model (The KITTI LiDAR coordinate system, i.e., x: forward, y: left, z: top, the origin is at about 1.6m above the ground plane).

closed 2020-05-07 by KevinDu1

Manueljohnson063, 2021-04-30:
@KevinDu1 hai sir did you solved the issue

yun9993, 2022-01-27:
> 

> 输入点云的坐标系是什么？应与提供的模型相同（KITTI LiDAR 坐标系，即x：前，y：左，z：上，原点在地平面以上约1.6m）。

How to modify it?