open-mmlab/OpenPCDet #29: Problem installing pcdet library

MYSystemLogistics, 2020-04-05:
Hi @sshaoshuai,

I'm trying to install pcdet, but when I run `python setup.py develop` I get this error:
```
  File "setup.py", line 26
    print('__version__ = "%s"' % version, file=f)
                                              ^
SyntaxError: invalid syntax
```
How can I solve it?

Thank you for your attention.

sshaoshuai, 2020-04-06:
Do you use python3 for this repo? If not, please try to use python3.

sshaoshuai, 2020-04-07:
Have you checked your `spconv` version?  This repo use `spconv v1.0` ([commit 8da6f96](https://github.com/traveller59/spconv/tree/8da6f967fb9a054d8870c3515b1b44eca2103634)) currently.

closed 2020-04-24 by sshaoshuai

natsume7, 2021-05-06:
i get the same problem but  after Run cd ./dist, use pip to install generated whl file ,it work