open-mmlab/OpenPCDet #32: 关于旋转角 变换的困惑，求帮助

xzswater, 2020-04-07:
我们的坐标系跟kitti坐标系不一样（需要做转换），关于代码中旋转角度的细节，有些困惑，烦请赐教~

pcdet/models/rcnn/partA2_rcnn_net.py
[image]

[image]

sshaoshuai, 2020-04-07:
We have the following definitions: 
1. The LiDAR coordinate denotes the KITTI LiDAR coordinate system, with x: forward, y: left, z: up, and **the direction of zero orientation is along the -y**.
2. The local coordinate denotes the coordinate system of a object, with x: heading direction of this object, y: perpendicular to x, z: up, and **the direction of zero orientation is along the x**. 

Hence, for transforming the LiDAR coordinate to local coordinate,  we should rotate the object points with `ry + np.pi / 2`

xzswater, 2020-04-08:
@sshaoshuai  Thank you very much !  And, could you help to explan why dir_offset set to pi / 4 ? 

tools/cfgs/PartA2.yaml
[image]

pcdet/utils/box_coder_utils.py
[image]

closed 2020-05-20 by xzswater

penghao1990, 2020-06-10:
您好，请问为什么需要dir_offset，并且设置为pi/4呢？我不明白，能否帮助一下。谢谢！

DeclK, 2021-11-14:
I found someone else's [answer](https://github.com/open-mmlab/OpenPCDet/issues/80#issuecomment-641865296) might solve this question