open-mmlab/OpenPCDet #33: how to change config files to not use intensity features?

gujiaqivadin, 2020-04-07:
Hello, sshaoshuai!
     Thansk for sharing your great work! I have a problem, now I am working on a point cloud without intensity or concat a fake intensity. How to change pointpillars/second.yaml files to choose not to use intensity featues in the model training?

Thx a lot!!

chreisinger, 2020-04-07:
Change the value of NUM_POINT_FEATURES.use in the corresponding config file (pointpillars.yaml/second.yaml) from 4 to 3.
NUM_POINT_FEATURES.total gets used in the sampler.

sshaoshuai, 2020-04-07:
Yes, thanks for the explanation of @chreisinger. 
I use `NUM_POINT_FEATURES.total` to indicate the total number of features of each point (`[x, y, z, ...]`, the first three features should be the point coordinate), and `NUM_POINT_FEATURES.use` to indicate how many features (the first few features) we are using in the model. 
Note that currently this strategy only supports to use the first few features of each point in order.

closed 2020-04-08 by gujiaqivadin