open-mmlab/OpenPCDet #36: Reproduce the results in pedestrian and cyclist

memoiry, 2020-04-09:
Hi, shaoshuai.

I have tried to run both PartA2_fc and PartA2_fc and get similar results in both car, pedestrian and cyclist classes, as follows.

```
 Car AP@0.70, 0.70, 0.70:
bbox AP:95.7255, 89.4638, 89.1119
bev  AP:90.2068, 87.8669, 87.2665
3d   AP:89.3621, 79.2417, 78.5225
aos  AP:95.70, 89.38, 88.96
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:98.3415, 93.9843, 91.9963
bev  AP:94.8540, 88.7180, 88.4109
3d   AP:92.0074, 82.7198, 80.3088
aos  AP:98.31, 93.87, 91.82
Car AP@0.70, 0.50, 0.50:
bbox AP:95.7255, 89.4638, 89.1119
bev  AP:95.7246, 89.4733, 89.2078
3d   AP:95.6730, 89.4277, 89.1581
aos  AP:95.70, 89.38, 88.96
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:98.3415, 93.9843, 91.9963
bev  AP:98.3214, 94.4235, 94.1939
3d   AP:98.2968, 94.3267, 93.9227
aos  AP:98.31, 93.87, 91.82
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:70.7935, 67.2700, 63.9463
bev  AP:65.4293, 58.6606, 54.8699
3d   AP:61.8364, 56.1250, 49.9829
aos  AP:67.70, 63.70, 59.98
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:72.1216, 67.7394, 63.4741
bev  AP:65.3724, 58.7195, 53.6765
3d   AP:61.7966, 55.4602, 49.3486
aos  AP:68.66, 63.85, 59.09
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:70.7935, 67.2700, 63.9463
bev  AP:76.2861, 72.4894, 67.8042
3d   AP:76.2115, 72.3084, 67.6083
aos  AP:67.70, 63.70, 59.98
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:72.1216, 67.7394, 63.4741
bev  AP:77.3366, 73.6643, 68.9374
3d   AP:77.2502, 73.4974, 68.7337
aos  AP:68.66, 63.85, 59.09
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:91.6235, 80.4274, 76.0859
bev  AP:87.3458, 73.9558, 70.8896
3d   AP:85.9091, 71.2065, 68.1585
aos  AP:91.52, 80.08, 75.71
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:94.3850, 81.4298, 77.3967
bev  AP:91.6213, 74.9996, 71.4178
3d   AP:89.9500, 71.3220, 67.8678
aos  AP:94.26, 81.04, 77.02
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:91.6235, 80.4274, 76.0859
bev  AP:90.6499, 77.5425, 73.2080
3d   AP:90.6464, 77.5408, 73.2018
aos  AP:91.52, 80.08, 75.71
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:94.3850, 81.4298, 77.3967
bev  AP:94.0451, 78.3774, 74.3371
3d   AP:94.0441, 78.3747, 74.3336
aos  AP:94.26, 81.04, 77.02
```

Results on Car are good but results on Pedestrian (61.7966, 55.4602, 49.3486) and Cyclist (85.9091, 71.2065, 68.1585) have a big gap compared to results in the paper, which is (70.37, 63.84, 57.48) on Ped and (88.31, 73.07, 70.20) for Cyclist, especially for the Ped.

Could you please suggest a reason for the degradation?

memoiry, 2020-04-09:
Update:

I also test the model `PartA2_fc` provided in the README, which gives a similar result.

 (62.8704, 56.3679, 50.1842) on Ped and (86.9893, 71.3273, 68.3141) on Cyclist.

sshaoshuai, 2020-04-10:
This results is as expected for 3 classes of a the single model. Training pedestrian/cyclist with separate models and also modifying the config  generally achieve better results. Also the performance of pedestrian/cyclist is not stable on KITTI dataset, and the released codes have been highly refactored and I mainly focus on car objects to tune the performance.

memoiry, 2020-04-10:
@sshaoshuai Thanks for your reply!

Could you please suggest how to modify the config file to reproduce the result in the paper except training in a separate model for ped/cyc?

sshaoshuai, 2020-04-10:
Actually the cyclist is easier than pedestrian, I just try to train the cyclist with more times, and the performances between training cyclist with a single model or joint with car are similar. For the pedestrian, you could also try to upsample the BEV feature map to 4x for putting anchors as pedestrain has smaller size. Finally, I found the fact was that the anchor-free version (PartA2-free which generates proposals like PointRCNN) achieves better and more stable results on pedestrian/cyclist on KITTI dataset.

closed 2020-04-17 by sshaoshuai