open-mmlab/OpenPCDet #40: How to get the index of the intermediate layer

519830100, 2020-04-11:
Thanks for sharing the code. I want to implement a one-stage bounding box predictor using the bottleneck layer. But I don't know how to get the corresponding regression gt of the bottleneck layer, x_conv4 layer for example. Can you provide some suggestions?

sshaoshuai, 2020-04-14:
What do you mean by "index of intermediate layer"? 
You could use x_conv4.indices to get the indices of each non-empty voxel.

519830100, 2020-04-14:
@sshaoshuai Here is the thing, I want to build an 8s output prediction, which means I have to get the gt of 8s. Given the 1s gt, how to find the corresponding map between 8s and 1s, i.e. how to get the downsampled ground truth.

sshaoshuai, 2020-04-17:
The answer varies based on the detailed setting (such as how do you use the GT), maybe you could also consider to upsample voxel to the 1s.

closed 2020-04-24 by sshaoshuai