open-mmlab/OpenPCDet #41: Some problem about camera coordinates 3D box IoU compute

GangZhang842, 2020-04-11:
Hi, recently i use "d3_box_overlap" in eval.py, to compute the same two 3D boxes 3D IoU. The IoU should be 1.0, but i get 8.3721865e-09. Some thing must be wrong，but i don`t know why.

sshaoshuai, 2020-04-14:
There is a small bug about computing the 3D IoU of two exactly same boxes. I didn't fix it since this code is borrowed from [traveller59/kitti-object-eval-python](https://github.com/traveller59/kitti-object-eval-python) and this bug is seldom triggered. 
You could try to modify here to fix it: https://github.com/sshaoshuai/PCDet/blob/master/pcdet/datasets/kitti/kitti_object_eval_python/eval.py#L131

GangZhang842, 2020-04-15:
Thank you very much!

closed 2020-04-17 by sshaoshuai

qifeng22, 2023-10-18:
hello, do you konw how to modify?

> There is a small bug about computing the 3D IoU of two exactly same boxes. I didn't fix it since this code is borrowed from [...]