open-mmlab/OpenPCDet #42: Use PartA2_car.yaml and PartA2.yaml to inference

shan-ying, 2020-04-14:
Thanks for releasing such useful package.
I have some question when I tried to inference the KITTI Data

The first question: 

 I run  
python ./tools/test.py --cfg_file ./cfgs/PartA2_car.yaml --batch_size 2 --ckpt ./weight/PartA2.pth --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True

I can get this result 
[image]

But when I run  ./tools/test.py --cfg_file ./cfgs/PartA2.yaml --batch_size 2 --ckpt ./weight/PartA2.pth --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True

[image]

It seems like have some problem on loading weight?
And How can I use the PartA2.yaml to inference KITTI data?

The second question:
In PartA2.yaml the model name is UNetV2, but I when I change the command 
--set MODEL.RPN.BACKBONE.NAME UNetV2 
[image]

Also get the problem about weight.
I don't know how to use UNetV2.

Can you provide some suggestion? Thank you!

sshaoshuai, 2020-04-14:
For question 1: The config of the provided `PartA2.pth` (I have renamed it to `PartA2_car.pth` to avoid ambiguity) if `PartA2_car.yaml`, and you could not use `PartA2.yaml` to load it.
For question 2: For training with any config, you don't need to add `--set`, just use the default config.

shan-ying, 2020-04-16:
Thank you very much! :)

closed 2020-04-16 by shan-ying