open-mmlab/OpenPCDet #45: error when evaluating PartA2_car model

amiltonwong, 2020-04-17:
Hi, @sshaoshuai ,

I got the following error when I perform evaluation by the following command:
`python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2_car.pth --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True`

```
2020-04-17 09:51:06,796   INFO  ==> Done (loaded 357/357)
2020-04-17 09:51:06,847   INFO  *************** EPOCH 2 EVALUATION *****************
eval:   0%|                                                     | 0/943 [00:00<?, ?it/s]Traceback (most recent call last):
  File "test.py", line 181, in <module>
    main()
  File "test.py", line 177, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id)
  File "test.py", line 59, in eval_single_ckpt
    model, test_loader, epoch_id, logger, result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/data/code10/PCDet/tools/eval_utils/eval_utils.py", line 46, in eval_one_epoch
    pred_dicts, ret_dict = model(input_dict)
  File "/root/anaconda3/envs/pytorch1.3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 541, in __call__
    result = self.forward(*input, **kwargs)
  File "/data/code10/PCDet/pcdet/models/detectors/PartA2_net.py", line 112, in forward
    batch_size, voxel_centers, coords, rpn_ret_dict, input_dict
  File "/data/code10/PCDet/pcdet/models/detectors/PartA2_net.py", line 77, in forward_rcnn
    use_binary_dir_classifier=cfg.MODEL.RPN.RPN_HEAD.ARGS.get('use_binary_dir_classifier', False)
  File "/data/code10/PCDet/pcdet/utils/box_coder_utils.py", line 129, in decode_with_head_direction_torch
    opp_labels = (batch_box_preds[..., -1] > 0) ^ dir_labels.byte()
RuntimeError: Expected object of scalar type Bool but got scalar type Byte for argument #2 'other' in call to _th_xor
eval:   0%|                                                     | 0/943 [00:08<?, ?it/s]
(pytorch1.3) root@Lab-PC:/data/code10/PCDet/tools
```

The detailed log is [here](https://gist.githubusercontent.com/amiltonwong/679dfa452ccfc3724c09c734f81503e3/raw/4f4d329d4120699dd3440facaafc870d04fae99c/log.txt).

Any hints to solve this issue? 
THX!

asadnorouzi, 2020-04-17:
@amiltonwong I had the same issue which was solved by changing line 129 in `box_coder_utils.py` as below:
```
opp_labels = (batch_box_preds[..., -1] > 0) ^ dir_labels.bool()
```

I think it's just due to different pytorch version.

amiltonwong, 2020-04-17:
@asadnorouzi , thanks a lot for your suggestion. It works after modified :)

closed 2020-04-17 by amiltonwong