open-mmlab/OpenPCDet #46: CUDA error: invalid configuration argument

chayangkultan96, 2020-04-23:
I have been getting the following error when I try to train. It only happens when I try to limit classes I train on.

> Traceback (most recent call last):
>   File "train.py", line 155, in <module>
>     main()
>   File "train.py", line 148, in main
>     max_ckpt_save_num=args.max_ckpt_save_num
>   File "/s/dat/UserFolders/ctan24/PCDet-master/tools/train_utils/train_utils.py", line 81, in train_model
>     leave_pbar=(cur_epoch + 1 == total_epochs)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/tools/train_utils/train_utils.py", line 36, in train_one_epoch
>     loss, tb_dict, disp_dict = model_func(model, batch)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/models/__init__.py", line 25, in model_func
>     ret_dict, tb_dict, disp_dict = model(input_dict)
>   File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
>     result = self.forward(*input, **kwargs)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/models/detectors/PartA2_net.py", line 112, in forward
>     batch_size, voxel_centers, coords, rpn_ret_dict, input_dict
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/models/detectors/PartA2_net.py", line 98, in forward_rcnn
>     rcnn_ret_dict = self.rcnn_net.forward(rcnn_input_dict)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/models/rcnn/partA2_rcnn_net.py", line 323, in forward
>     targets_dict = self.assign_targets(batch_size, rcnn_dict)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/models/rcnn/partA2_rcnn_net.py", line 27, in assign_targets
>     targets_dict = proposal_target_layer(rcnn_dict, roi_sampler_cfg=self.rcnn_target_config)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/models/model_utils/proposal_target_layer.py", line 14, in proposal_target_layer
>     sample_rois_for_rcnn(rois, gt_boxes, roi_raw_scores, roi_labels, roi_sampler_cfg)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/models/model_utils/proposal_target_layer.py", line 77, in sample_rois_for_rcnn
>     iou3d = iou3d_nms_utils.boxes_iou3d_gpu(cur_roi, cur_gt[:, 0:7])  # (M, N)
>   File "/s/dat/UserFolders/ctan24/PCDet-master/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 52, in boxes_iou3d_gpu
>     vol_a = (boxes_a[:, 3] * boxes_a[:, 4] * boxes_a[:, 5]).view(-1, 1)
> RuntimeError: CUDA error: invalid configuration argument

sshaoshuai, 2020-04-24:
Could you print out the shape of cur_roi and cur_gt when it happens?

lan-sky, 2020-05-12:
do you solve this problem？

closed 2020-05-17 by chayangkultan96

lan-sky, 2020-05-19:
hey，How did you solve the problem

chayangkul99, 2020-05-19:
I turns out some of the scene did not have any GT labels, I was looking at the wrong batch in my other issue thread. So just print out the cur_roi and cur_gt as suggested by sshaoshuai and debug from there. If it is the same issue, update the dataloader to exclude scenes without any GT labels.