open-mmlab/OpenPCDet #49: spconv1.1 possible?

monghimng, 2020-04-26:
Hi, thank you for the amazing work. I tried installing spconv1., but their cuda libs search locations and their build process kept breaking on my lab's servers. I installed spconv1.1 instead, and it just worked. However, I noticed in several places this repo stresses spconv1.0. I am wondering if there is anything that absolutely needs spconv1.0.

The 2 main places this repo uses spconv is the `voxel_generator` and the convolutions.

sshaoshuai, 2020-04-26:
Please refer to here https://github.com/sshaoshuai/PCDet/pull/23 for the issue about `spconv 1.1`, where you could also find a `spconv 1.1` version supported by @codyreading at https://github.com/codyreading/PCDet/tree/spconv-1.1.

sshaoshuai, 2020-04-27:
Now `PCDet` has supported `spconv 1.1` as modified here https://github.com/sshaoshuai/PCDet/commit/2244be47d3721ba95cf812c8646ffa305b153db4.

closed 2020-05-11 by sshaoshuai

MartinHahner88, 2020-05-19:
> Hi, thank you for the amazing work. I tried installing spconv1., but their cuda libs search locations and their build process kept breaking on my lab's servers. I installed spconv1.1 instead, and it [...]

I am stuck with the same issue, 
I just cannot achieve a successful spconv v1.0 build on our lab's servers.
@monghimng: Were you finally able to build spconv v1.0 ?

monghimng, 2020-05-19:
I just installed v1.1 and the branch the author provided. It somehow worked for me.

codyreading, 2020-05-19:
The repo has support for spconv1.1, the main difference is the GPU memory usage of the PartA2 ( a lot larger in spconv==1.1). Still should work fine from https://github.com/sshaoshuai/PCDet/commit/2244be47d3721ba95cf812c8646ffa305b153db4.

MartinHahner88, 2020-05-19:
> I just installed v1.1 and the branch the author provided. It somehow worked for me.

@monghimng: 
That's great news, if you are using conda, could you please provide the output of `conda list`.
Then I could maybe just "clone" your conda environment and all is good.

@codyreading: 
Thank you for pointing out the different GPU memory usage. 
I tried both _SECOND_ and _PartA^2Net_ with `batch_size=1` 
on a GeForce RTX 2080 Ti with 11GB and also on a TITAN Xp with 12GB GPU memory
and the training in both cases still crashed with `unknown device type` as mentioned in issue [#70](https://github.com/sshaoshuai/PCDet/issues/70).

I still think the issue most likely comes from my specific set of versions for 
- python 
- pytorch 
- CUDA
- cudnn. 

that I am using. For all details please see issue [#70](https://github.com/sshaoshuai/PCDet/issues/70).

Sadly, I found no other combination yet that allowed both, 
a successful build of spconv **and** a successful installation of PCDet.

codyreading, 2020-05-19:
Hi @MartinHahner88,

Doesn't look like I've seen that issue. I've been using the following versions which has worked for me, and using `pip` with `virtualenv` to install the packages.

- Python: 3.7
- Torch: 1.4
- CUDA: 10.1
- CUDNN: 7.6.5

MartinHahner88, 2020-05-20:
@codyreading 
> Doesn't look like I've seen that issue. I've been using the following versions which has worked for me, and using `pip` with `virtualenv` to install the packages.
> 
>     * Python: 3.7
> 
>     * Torch: 1.4
> 
>     * CUDA: 10.1
> 
>     * CUDNN: 7.6.5

Those versions, unfortunately, let to yet another issue for me, see [here](https://github.com/sshaoshuai/PCDet/issues/70#issuecomment-631675659).