open-mmlab/OpenPCDet #50: Expected behavior when there are no objects in ground truth

vatsal-shah, 2020-04-28:
I am trying to run the pretrained model on the val split of the Nuscenes dataset. I have converted it to the KITTI format using their script. 
Some frames in the Nuscenes val set contain 0 objects, and hence have no ground truth annotations. I wanted to know what the expected behavior should be in this case. When I run
```
python kitti_dataset.py create_kitti_infos
```
I run into the following error when there are no objects in the ground truth file.

```
File "kitti_dataset.py", line 510, in <module>
    save_path=cfg.ROOT_DIR / 'data' / 'kitti'
  File "kitti_dataset.py", line 484, in create_kitti_infos
    kitti_infos_val = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "kitti_dataset.py", line 164, in get_infos
    info = process_single_scene(sampl_id)
  File "kitti_dataset.py", line 113, in process_single_scene
    annotations['bbox'] = np.concatenate([obj.box2d.reshape(1, 4) for obj in obj_list], axis=0)
  File "<__array_function__ internals>", line 6, in concatenate
ValueError: need at least one array to concatenate
``` 

Could you let me know what should be done in this case when there are no objects in the ground truth?

ltphy, 2020-05-04:
in dataset.py, when preparing data, the ground truth boxes are filtered with point cloud range. You may comment this. 
Or do a clean-up with boxes outside the pointcloud range when create a pickle train file.

PengfeiMa, 2020-09-17:
Hi, @vatsal-shah, I met the same error as yours. I also Nuscenes  to the KITTI format using their script. 
But I found that there are some label file is empty in train set and val set, just del the empty label file name from the ImageSet/train.txt or val.txt.
It works!

sshaoshuai, 2020-11-04:
We will re-sample a scene if we find there doesn't exist any objects in current scene.  See https://github.com/open-mmlab/OpenPCDet/pull/340.

closed 2020-11-04 by sshaoshuai

GYE19970220, 2021-01-19:
> Hi, @vatsal-shah, I met the same error as yours. I also Nuscenes to the KITTI format using their script.
> But I found that there are some label file is empty in train set and val set, just del the empty label file name from the ImageSet/train.txt or val.txt.
> It works!

Hi @PengfeiMa ,I met the same error as yours, how to check if the label file is empty or not ?

PengfeiMa, 2021-01-19:
@GYE19970220 Open the label file, if there is nothing,  then delete the file name from the ImageSet/train.txt or val.txt.

GYE19970220, 2021-01-19:
@PengfeiMa Thanks, bro. I have found the empty label. Del them and train other problem occurs,I will keep on going.Can you leave a mailbox to facilitate communication？Mine 912236521@qq.com

maxiuw, 2023-11-09:
As nonsense as it is, I didn't find any other way than really remove it from the train.txt file. Additionally it is not super easy to identify which label is wrong since it does not crash right away. My brute force fix is adding to `get_infos` loop this 
```
 try:
                    annotations['bbox'] = np.concatenate([obj.box2d.reshape(1, 4) for obj in obj_list], axis=0)
                except:
                    print(sample_idx)
                    
                    file = open("/home/raghav/code/UDA/DA-IA-SSD/data/kitti/ImageSets/val.txt", "r")
                    lines = file.readlines()
                    new_lines = []
                    for line in lines:
                        if sample_idx not in line.strip():
                            new_lines.append(line)
                    file.close()
                    file = open("/home/raghav/code/UDA/DA-IA-SSD/data/kitti/ImageSets/val.txt", "w")
                    file.writelines(new_lines)
                    file.close()
```