open-mmlab/OpenPCDet #51: IndexError: index 3 is out of bounds for axis 0 with size 3

nmll, 2020-04-28:
Traceback (most recent call last):
  File "test.py", line 188, in <module>
    main()
  File "test.py", line 184, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id)
  File "test.py", line 63, in eval_single_ckpt
    model, test_loader, epoch_id, logger, result_dir=eval_output_dir, save_to_file=args.save_to_file
at kitt_dataset.py-dataset_init-index=16
  File "/share/lycode/PCDet/tools/eval_utils/eval_utils.py", line 50, in eval_one_epoch
at kitt_dataset.py-dataset_init-index=24
    for i, data in enumerate(dataloader):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 582, in __next__
    return self._process_next_batch(batch)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 608, in _process_next_batch
    raise batch.exc_type(batch.exc_msg)
IndexError: Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/worker.py", line 99, in _worker_loop
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/worker.py", line 99, in <listcomp>
    samples = collate_fn([dataset[i] for i in batch_indices])
  File "/share/lycode/PCDet/pcdet/datasets/kitti/kitti_dataset.py", line 451, in __getitem__
    annos = common_utils.drop_info_with_name(annos, name='DontCare')
  File "/share/lycode/PCDet/pcdet/utils/common_utils.py", line 76, in drop_info_with_name
    ret_info[key] = info[key][keep_indices]
IndexError: index 3 is out of bounds for axis 0 with size 3

eval:   0%|                                                                                     | 0/2002 [00:00<?, ?it/s]

Could I ask about this error when i run CUDA_VISIBLE_DEVICES=2 python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt cfgs/PartA2_car.pth --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True?I don't know how to solve it ,Thanks for help!

sshaoshuai, 2020-05-06:
Have you followed the README to run this command? 
If so, could you please print out the key and key_indices for this error?

nmll, 2020-05-07:
yes,thanks,I have solved this.the reason is my label file 'dontcare' is not structured

closed 2020-05-07 by sshaoshuai

sarimmehdi, 2020-07-29:
Hello. I am also getting this error. Can you please explain how you fixed this?

nmll, 2020-07-30:
@sarimmehdi  I have solved this because my dataset have different format with kitti.maybe you have to check you dataset.

sarimmehdi, 2020-07-30:
I have the same dataset as kitti

yustaub, 2020-10-22:
> I have the same dataset as kitti

may I ask have you solved this problem?

nmll, 2020-10-22:
I forgot a lot,maybe is some label file do not have dontcare raw

yustaub, 2020-10-22:
> I forgot a lot,maybe is some label file do not have dontcare raw

ok, I will check data, thanks