open-mmlab/OpenPCDet #55: confused with your anchor assign?

AndyYuan96, 2020-05-05:
Hi, shaoshuai, I'm a little confused with your anchor assign in PCDet/pcdet/models/bbox_heads/anchor_target_assigner.py : 
in create_target_np():
 `        gt_inds_force = anchor_to_gt_argmax[anchors_with_max_overlap]
            labels[anchors_with_max_overlap] = gt_classes[gt_inds_force]
            gt_ids[anchors_with_max_overlap] = gt_inds_force
            # Fg label: above threshold IOU
            pos_inds = anchor_to_gt_max >= matched_threshold
            gt_inds = anchor_to_gt_argmax[pos_inds]
            labels[pos_inds] = gt_classes[gt_inds]
            gt_ids[pos_inds] = gt_inds
            bg_inds = np.where(anchor_to_gt_max < unmatched_threshold)[0]
`

anchors_with_max_overlap : anchor ids whose iou with a gt is biggest among all anchors with the gt.
anchor_to_gt_argmax: gt ids: for each anchor, the anchor has biggest iou with the gt among all gt.
the question is:
    gt_inds_force = anchor_to_gt_argmax[anchors_with_max_overlap]
    you want to use anchors_with_max_overlap find the gt id, and use the id as label.
    setting: 5 anchor, 5 gt.
    gt_1 has biggest iou with anchor_1, anchor_1 may have biggest iou with gt_2.
    you want to use anchor_to_gt_argmax[anchors_with_max_overlap] to get gt_1 , but anchor_to_gt_argmax[anchors_with_max_overlap] give you gt_2.

sshaoshuai, 2020-05-06:
Hi, 
It seems that your mentioned situation does exist, but I think it may not affect too much on the final performance since assigning anchor_1 to either gt_1 or gt_2 is reasonable.

MartinHahner88, 2020-05-19:
@AndyYuan96: 
> gt_1 has biggest iou with anchor_1 among all anchors, anchor_1 may have biggest iou with gt_2 among all gts 

I agree with @sshaoshuai and think that this exceptional situation will rarely happen since the two 3D bounding boxes of gt_1 and gt_2 cannot overlap in 3D space (contrary to two 2D bounding boxes in 2D images).

closed 2020-06-02 by sshaoshuai