open-mmlab/OpenPCDet #56: Make my own 3d dataset

nicholasguimaraes, 2020-05-05:
Hey sshaoshuai, I was looking for more information on how the KITTI 3d dataset was made.
Found good information regarding all of the parameters and data needed to train the network here;

https://github.com/yanii/kitti-pcl/blob/master/KITTI_README.TXT

http://www.cvlibs.net/publications/Geiger2013IJRR.pdf

I wonder if you know of any repository that could have the codes to generate the parameters to the calibration file.

Regarding annotations (bounding box) do I need to annotate both the 2d rgb img and the 3d point cloud?

sshaoshuai, 2020-05-08:
Sorry, I could not help much for these questions. Maybe you could find more related materials under the repo of other 3D detection datasets like NuScenes and Waymo.

neverrop, 2020-05-19:
@nicholasguimaraes have you found any methods for this? It'll be so nice if you can share this.

closed 2020-07-17 by sshaoshuai

zzqjh, 2020-09-22:
hello i also want to use my own datasat .Do you have used your own dataset for the model? could you please give some suggestions for me ？