open-mmlab/OpenPCDet #58: What does the road planes for data augmentation do?

Hastyrush, 2020-05-08:
Hello, you mentioned that road planes can be downloaded additionally for data augmentation of the KITTI 3D Object detection dataset during training. May I know what does the augmentation actually mean or do? Does it remove the road plane or transforms it in a certain way? And how does it help with the training of the model.

Thanks a lot!
[image]

sshaoshuai, 2020-05-08:
The road planes are used for GT sampling augmentation, where we could 'paste' the additional GTs to the road plane of current scene.

Hastyrush, 2020-05-08:
I see. Thanks a lot for the information! Will be closing this issue.

closed 2020-05-08 by Hastyrush

Petros626, 2024-10-16:
@sshaoshuai can you show me in the code where the gt_database .bin files are loaded to be gt_sampled during training? And where the safety checks are made, that there is no overlapping with other objects.

Thank you very much!