open-mmlab/OpenPCDet #59: any plan for waymo and MVF?

AndyYuan96, 2020-05-10:
Hi, shaoshuai, thank you for provide so Clear code structure,  it did helps me finding why my implementation 's result is poor than official. 
do you have any plan for implementing waymo dataloader recently, if not I will give it a try within a mouth. and also I can add my implementation of MVF and pixor to your PCDet.

sshaoshuai, 2020-05-11:
Hi AndyYuan, thanks for your interests of this repo. I plan to update the PCDet repo and release the waymo related dataloader and models next month.

closed 2020-05-12 by AndyYuan96

Saurabh7, 2020-06-05:
Hi @sshaoshuai 
Just wondering if there are any updates regarding the waymo support ?
or pointers on how to adapt for waymo, should we follow the template mentioned here: https://github.com/sshaoshuai/PCDet/blob/master/pcdet/datasets/dataset.py ?

tianweiy, 2020-06-06:
Also looking forward to Waymo release.

ideasplus, 2022-05-14:
@AndyYuan96 Hello, I really want to know where can find your implementation of pixor based on OpenPCDet? I haven't found a correct public implementation of pixor yet. I need to do some experiments based on pixor. Looking forward to your reply!