open-mmlab/OpenPCDet #60: Reduce point

youyousunny, 2020-05-11:
hello, i have noticed that there is an operation about reduction points in KITTI data in the algorithm of SECOND. If we do some openration about reduction points in KITTI in PCdet alogirithm, the precision of MAP maybe increased?

jiehaohuang, 2020-05-18:
when testing a model of 'second' with provided second.pth, it gets a result that is 
 Car AP@0.70, 0.70, 0.70:
bbox AP:30.4553, 28.1927, 27.1099
bev  AP:30.1152, 27.8749, 26.7538
3d   AP:29.9210, 27.1801, 25.8764
aos  AP:30.44, 28.15, 27.05
......
......
......
I want to ask that If I make any mistakes?

MartinHahner88, 2020-05-19:
@youyousunny 
> hello, i have noticed that there is an operation about reduction points in KITTI data in the algorithm of SECOND. If we do some openration about reduction points in KITTI in PCdet alogirithm, the precision of MAP maybe increased?

I think the point cloud reduction in PCDet is handled "on the fly" by [`get_fov_flags`](https://github.com/sshaoshuai/PCDet/blob/2244be47d3721ba95cf812c8646ffa305b153db4/pcdet/datasets/kitti/kitti_dataset.py#L69)
and setting `FOV_POINTS_ONLY: True` as in [tools/cfgs/second.yaml](https://github.com/sshaoshuai/PCDet/blob/2244be47d3721ba95cf812c8646ffa305b153db4/tools/cfgs/second.yaml#L6) for example.

See [line 430](https://github.com/sshaoshuai/PCDet/blob/2244be47d3721ba95cf812c8646ffa305b153db4/pcdet/datasets/kitti/kitti_dataset.py#L430) in `kitti_dataset.py`.

So there is no need to create a `reduced_pointcloud` prior to training.

closed 2020-06-02 by sshaoshuai