open-mmlab/OpenPCDet #62: interesting result?

AndyYuan96, 2020-05-12:
Hi, shaoshuai, did you try to not use ry as the angle label ? I convert ry to -(ry + pi / 2) to convert ry to lidar coordinate, and then use -(ry + pi / 2) as angle label, after finishing training, I convert it back to ry.
The result is very interesting,  -(ry + pi / 2) is poor than ry when iou is 0.7, generally lower than 20 points, but when the iou is 0.5, the result is close.

tianweiy, 2020-05-12:
@AndyYuan96  Why do you want to convert it to -(ry + pi/2)? I am not very familiar with pcdet, but the reason [SECOND](https://github.com/traveller59/second.pytorch) and [Det3D](https://github.com/poodarchu/det3d) converts it to -(ry + pi/2) for nuScenes is because nuScenes uses a different coordinate system than KITTI. However, for PCDet, the conversion from an original KITTI label to -(ry + pi/2) may break those augmentation / nms / other code, one example is the flip augmentation I guess.

AndyYuan96, 2020-05-13:
@tianweiy ,in KITTI,  -(ry + pi/2) is the true angle of bounding box in lidar coordinate, convert to -(ry+pi/2) will not break nms and other code, except for augumentation. I use -(ry + pi/2) , because the angle is the true angle of ground truth bounding box in lidar coordinate, and the anchor is also in lidar coordinate. If use ry to assign gt with anchor, we are using gt(in camera coordinate) and anchor(in lidar coordinate) to match,  though it's a little confused to understand.  but it does work.

closed 2020-05-13 by AndyYuan96