open-mmlab/OpenPCDet #63: Generate KITTI data infos error

zhanghming, 2020-05-14:
Hi, thanks for your great work.

When I execute `python kitti_dataset.py create_kitti_infos`, I get following problem:
```shell
Traceback (most recent call last):
  File "pcdet/datasets/kitti/kitti_dataset.py", line 518, in <module>
    save_path=cfg.ROOT_DIR / 'data' / 'kitti'
  File "pcdet/datasets/kitti/kitti_dataset.py", line 491, in create_kitti_infos
    with open(val_filename, 'wb') as f:
TypeError: invalid file: PosixPath('/mnt/study/Research/code/PCDet/data/kitti/kitti_infos_val.pkl')
```
Looks like the `PosixPath` has some problems, and I also get the problem like:
```shell
TypeError: join() argument must be str or bytes, not 'PosixPath'
```
When code run into
```shell
self.root_split_path = os.path.join(self.root_path, 'training' if split != 'test' else 'testing')
```
in line 22, and after I replace this line with the hard coded KITTI dataset path in my computer, it became OK.

I cann't find any clues about these problems.

chreisinger, 2020-05-14:
Hi, 
`os.path.join(...)` requires type str as pointed out by the error message. This can be fixed with `str(yourPosixPath)`. I am not sure about the invalid file error.

zhanghming, 2020-05-15:
Thanks for your reply, I solved the problem by following your suggestions.

closed 2020-05-15 by zhanghming