open-mmlab/OpenPCDet #64: can someone please provide version numbers that work

MartinHahner88, 2020-05-15:
Can someone please provide version numbers of the following packages?

Python: ?
PyTorch: ?
CUDA: ?
CUDNN: ?
spconv: ?

I am using the following conda environment and run into the error below when I execute
`python setup.py develop`

```
# packages in environment at /home/mhahner/scratch/apps/anaconda3/envs/pcdet:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                      1_llvm    conda-forge
blas                      1.0                         mkl
bzip2                     1.0.8                h516909a_2    conda-forge
ca-certificates           2020.1.1                      0
certifi                   2020.4.5.1               py38_0
cmake                     3.17.0               h28c56e5_0    conda-forge
cudatoolkit               10.1.243             h6bb024c_0
cudatoolkit-dev           10.1.243             h516909a_3    conda-forge
cudnn                     7.6.5                cuda10.1_0
cycler                    0.10.0                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
easydict                  1.9                      pypi_0    pypi
expat                     2.2.9                he1b5a44_2    conda-forge
future                    0.18.2                   pypi_0    pypi
imagecodecs               2020.2.18                pypi_0    pypi
imageio                   2.8.0                    pypi_0    pypi
kiwisolver                1.2.0                    pypi_0    pypi
krb5                      1.17.1               h2fd8d38_0    conda-forge
ld_impl_linux-64          2.34                 h53a641e_0    conda-forge
libcurl                   7.69.1               hf7181ac_0    conda-forge
libedit                   3.1.20170329      hf8c457e_1001    conda-forge
libffi                    3.2.1             he1b5a44_1007    conda-forge
libgcc-ng                 9.2.0                h24d8f2e_2    conda-forge
libgfortran-ng            7.3.0                hdf63c60_5    conda-forge
libssh2                   1.8.2                h22169c7_2    conda-forge
libstdcxx-ng              9.2.0                hdf63c60_2    conda-forge
libuv                     1.34.0               h516909a_0    conda-forge
llvm-openmp               10.0.0               hc9558a2_0    conda-forge
llvmlite                  0.32.1                   pypi_0    pypi
matplotlib                3.2.1                    pypi_0    pypi
mkl                       2019.5                      281    conda-forge
mkl-service               2.3.0            py38h516909a_0    conda-forge
mkl_fft                   1.1.0            py38hc1659b7_1    conda-forge
mkl_random                1.1.0            py38h962f231_0
ncurses                   6.1               hf484d3e_1002    conda-forge
networkx                  2.4                      pypi_0    pypi
ninja                     1.10.0               hc9558a2_0    conda-forge
numba                     0.49.1                   pypi_0    pypi
numpy                     1.18.1           py38h4f9e942_0
numpy-base                1.18.1           py38hde5b4d6_1
openssl                   1.1.1g               h7b6447c_0
pillow                    7.1.2                    pypi_0    pypi
pip                       20.1               pyh9f0ad1d_0    conda-forge
protobuf                  3.11.3                   pypi_0    pypi
pyparsing                 2.4.7                    pypi_0    pypi
python                    3.8.2           he5300dc_7_cpython    conda-forge
python-dateutil           2.8.1                    pypi_0    pypi
python_abi                3.8                      1_cp38    conda-forge
pytorch                   1.5.0           py3.8_cuda10.1.243_cudnn7.6.3_0    pytorch
pywavelets                1.1.1                    pypi_0    pypi
pyyaml                    5.3.1                    pypi_0    pypi
readline                  8.0                  hf8c457e_0    conda-forge
rhash                     1.3.6             h14c3975_1001    conda-forge
scikit-image              0.17.2                   pypi_0    pypi
[... 397 lines cut ...]
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 87, in run
    _build_ext.run(self)
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 580, in build_extensions
    build_ext.build_extensions(self)
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 208, in build_extension
    _build_ext.build_extension(self, ext)
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 414, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1135, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/mhahner/scratch/apps/anaconda3/envs/pcdet/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1413, in _run_ninja_build
    raise RuntimeError(message)
RuntimeError: Error compiling objects for extension

MartinHahner88, 2020-05-15:
I have found a valid combination (see below).
I assume either python 3.8 or pytorch 1.5 caused the trouble

```
# packages in environment at /home/mhahner/scratch/apps/anaconda3/envs/apex:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                      1_llvm    conda-forge
bzip2                     1.0.8                h516909a_2    conda-forge
ca-certificates           2020.1.1                      0
certifi                   2020.4.5.1               py37_0
cmake                     3.17.0               h28c56e5_0    conda-forge
coloredlogs               14.0                     pypi_0    pypi
cudatoolkit               10.1.243             h6bb024c_0
cudatoolkit-dev           10.1.243             h516909a_3    conda-forge
cudnn                     7.6.5                cuda10.1_0
cycler                    0.10.0                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
easydict                  1.9                      pypi_0    pypi
expat                     2.2.9                he1b5a44_2    conda-forge
humanfriendly             8.2                      pypi_0    pypi
imagecodecs               2020.2.18                pypi_0    pypi
imageio                   2.8.0                    pypi_0    pypi
kiwisolver                1.2.0                    pypi_0    pypi
krb5                      1.17.1               h2fd8d38_0    conda-forge
libblas                   3.8.0               16_openblas    conda-forge
libcblas                  3.8.0               16_openblas    conda-forge
libcurl                   7.69.1               hf7181ac_0    conda-forge
libedit                   3.1.20181209         hc058e9b_0
libffi                    3.2.1             he1b5a44_1007    conda-forge
libgcc-ng                 9.2.0                h24d8f2e_2    conda-forge
libgfortran-ng            7.5.0                hdf63c60_6    conda-forge
liblapack                 3.8.0               16_openblas    conda-forge
libopenblas               0.3.9                h5ec1e0e_0    conda-forge
libssh2                   1.8.2                h22169c7_2    conda-forge
libstdcxx-ng              9.2.0                hdf63c60_2    conda-forge
libuv                     1.34.0               h516909a_0    conda-forge
llvm-openmp               10.0.0               hc9558a2_0    conda-forge
llvmlite                  0.32.1                   pypi_0    pypi
matplotlib                3.2.1                    pypi_0    pypi
mkl                       2020.1                      217    conda-forge
ncurses                   6.1               hf484d3e_1002    conda-forge
networkx                  2.4                      pypi_0    pypi
ninja                     1.10.0               hc9558a2_0    conda-forge
numba                     0.49.1                   pypi_0    pypi
numpy                     1.18.4           py37h8960a57_0    conda-forge
openssl                   1.1.1g               h7b6447c_0
pcdet                     0.1.0+2244be4             dev_0    <develop>
pillow                    7.1.2                    pypi_0    pypi
pip                       20.1               pyh9f0ad1d_0    conda-forge
protobuf                  3.11.3                   pypi_0    pypi
pyparsing                 2.4.7                    pypi_0    pypi
python                    3.7.5                h0371630_0
python-dateutil           2.8.1                    pypi_0    pypi
python_abi                3.7                     1_cp37m    conda-forge
pytorch                   1.4.0           py3.7_cuda10.1.243_cudnn7.6.3_0    pytorch
pywavelets                1.1.1                    pypi_0    pypi
pyyaml                    5.3.1                    pypi_0    pypi
readline                  7.0               hf8c457e_1001    conda-forge
rhash                     1.3.6             h14c3975_1001    conda-forge
scikit-image              0.17.2                   pypi_0    pypi
scipy                     1.4.1                    pypi_0    pypi
setuptools                46.3.0           py37hc8dfbb8_0    conda-forge
six                       1.14.0                   pypi_0    pypi
spconv                    1.1                      pypi_0    pypi
sqlite                    3.31.1               h7b6447c_0
tensorboardx              2.0                      pypi_0    pypi
tifffile                  2020.5.11                pypi_0    pypi
tk                        8.6.10               hed695b0_0    conda-forge
tqdm                      4.46.0                   pypi_0    pypi
wheel                     0.34.2                     py_1    conda-forge
xz                        5.2.5                h516909a_0    conda-forge
zlib                      1.2.11            h516909a_1006    conda-forge
```

MartinHahner88, 2020-06-05:
Finally SOLVED!

The final issue was, that **we tried to build spconv on a _Tesla K40_** GPU, 
which has a _Kepler_ architecture and is seemingly too old to build _spconv_.

_The reason why we always tried to build spconv on a Tesla K40 is that in our lab, it is hard to get a GPU on our GPU cluster interactively. (Usually, you can only submit jobs via qsub/SLURM.)
So we always tried to build spconv either on our local Linux clients or on old nodes which only have Tesla K40s. Then, out of despair, we tried to build spconv on a Titan X (Pascal) and we finally found a combination of requirements that worked._

So here is our solution:

```
conda create --name PCDet python=3.6 pytorch=1.1 cudatoolkit=9.2 cudatoolkit-dev=9.2 \
cmake --channel pytorch --channel=conda-forge
conda activate PCDet
conda install cudnn
conda install boost
git clone https://github.com/traveller59/spconv spconv_8da6f96 --recursive
cd spconv_8da6f96
git checkout 8da6f967fb9a054d8870c3515b1b44eca2103634
```

> If necessary: 
> Download and extract 
> [0001-Allow-to-specifiy-CUDA_ROOT-directory-and-pick-corre.patch.zip](https://github.com/sshaoshuai/PCDet/files/4737235/0001-Allow-to-specifiy-CUDA_ROOT-directory-and-pick-corre.patch.zip)
> and patch _spconv_ via:
> ```git am <PATH_TO_EXTRACTED_FILE>/0001-Allow-to-specifiy-CUDA_ROOT-directory-and-pick-corre.patch```

```
CUDA_ROOT=<PATH_TO_YOUR_CONDA_INSTALLATION>/conda_envs/PCDet python setup.py bdist_wheel
cd dist/
pip install *
```
Test _spconv_ via:
`python -c 'import spconv'`
(should just return and not raise any errors) 

```
cd ../..
git clone https://github.com/sshaoshuai/PCDet.git
cd PCDet/
pip install -r requirements.txt
CUDA_ROOT=<PATH_TO_YOUR_CONDA_INSTALLATION>/conda_envs/PCDet python setup.py develop
```

Done!

I hope these instructions help someone else who struggles to build _spconv_ as well.

closed 2020-06-05 by MartinHahner88

soheilAppear, 2020-08-18:
> CUDA_ROOT=<PATH_TO_YOUR_CONDA_INSTALLATION>/conda_envs/PCDet python setup.py bdist_wheel
> cd dist/
> pip install *

I have the issue with the above part. It gives me this error:

Release
-- The CUDA compiler identification is unknown
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - failed
-- Check for working CUDA compiler: /usr/bin/nvcc
-- Check for working CUDA compiler: /usr/bin/nvcc - broken
CMake Error at /home/soheil/anaconda3/envs/PCDet/share/cmake-3.18/Modules/CMakeTestCUDACompiler.cmake:52 (message):
  The CUDA compiler

    "/usr/bin/nvcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/soheil/OpenPCDet/spconv_8da6f96/build/temp.linux-x86_64-3.8/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_3aefc/fast && /usr/bin/make  -f CMakeFiles/cmTC_3aefc.dir/build.make CMakeFiles/cmTC_3aefc.dir/build
    make[1]: Entering directory '/home/soheil/OpenPCDet/spconv_8da6f96/build/temp.linux-x86_64-3.8/CMakeFiles/CMakeTmp'
    Building CUDA object CMakeFiles/cmTC_3aefc.dir/main.cu.o
    /usr/bin/nvcc    "--expt-relaxed-constexpr"   -c /home/soheil/OpenPCDet/spconv_8da6f96/build/temp.linux-x86_64-3.8/CMakeFiles/CMakeTmp/main.cu -o CMakeFiles/cmTC_3aefc.dir/main.cu.o
    ptxas fatal   : Value 'sm_20' is not defined for option 'gpu-name'
    CMakeFiles/cmTC_3aefc.dir/build.make:84: recipe for target 'CMakeFiles/cmTC_3aefc.dir/main.cu.o' failed
    make[1]: *** [CMakeFiles/cmTC_3aefc.dir/main.cu.o] Error 255
    make[1]: Leaving directory '/home/soheil/OpenPCDet/spconv_8da6f96/build/temp.linux-x86_64-3.8/CMakeFiles/CMakeTmp'
    Makefile:140: recipe for target 'cmTC_3aefc/fast' failed
    make: *** [cmTC_3aefc/fast] Error 2