open-mmlab/OpenPCDet #69: RuntimeError: CUDA error: invalid argument

zhanghming, 2020-05-16:
When I run
```shell
python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2_car.pth --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True
```
I got the following error:
```shell
2020-05-16 13:00:57,130   INFO  *************** EPOCH 2 EVALUATION *****************
eval:   0%|                                                                                                                                                                                                                                                       | 0/943 [00:00<?, ?it/s]Traceback (most recent call last):
  File "test.py", line 181, in <module>
    main()
  File "test.py", line 177, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id)
  File "test.py", line 59, in eval_single_ckpt
    model, test_loader, epoch_id, logger, result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/zhanghm/Programming/PCDet/PCDet/tools/eval_utils/eval_utils.py", line 46, in eval_one_epoch
    pred_dicts, ret_dict = model(input_dict)
  File "/home/zhanghm/.virtualenvs/pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zhanghm/Programming/PCDet/PCDet/pcdet/models/detectors/PartA2_net.py", line 106, in forward
    rpn_ret_dict = self.forward_rpn(**input_dict)
  File "/home/zhanghm/Programming/PCDet/PCDet/pcdet/models/detectors/PartA2_net.py", line 33, in forward_rpn
    **kwargs
  File "/home/zhanghm/.virtualenvs/pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zhanghm/Programming/PCDet/PCDet/pcdet/models/rpn/rpn_unet.py", line 278, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/zhanghm/.virtualenvs/pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zhanghm/.virtualenvs/pytorch/lib/python3.6/site-packages/spconv/modules.py", line 127, in forward
    input.features = module(input.features)
  File "/home/zhanghm/.virtualenvs/pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/zhanghm/.virtualenvs/pytorch/lib/python3.6/site-packages/torch/nn/modules/activation.py", line 99, in forward
    return F.relu(input, inplace=self.inplace)
  File "/home/zhanghm/.virtualenvs/pytorch/lib/python3.6/site-packages/torch/nn/functional.py", line 943, in relu
    result = torch.relu(input)
RuntimeError: CUDA error: invalid argument
eval:   0%| 
```
I cann't find any clues to solve this probelm.

Nvidia-Driver: 440.85
Cuda: 10.0

GuilinZ, 2020-05-20:
Hi, I met the same problem, have you solved this? Thanks! @zhanghming

sshaoshuai, 2020-06-03:
I could not figure out what happened here. What's the version of your `spconv`? And could you check out the status of the `input` for this line?

Zhoushanglin100, 2020-06-20:
I met exactly the same problem. Did you figure out how to solve this? Thx! @zhanghming

youyousunny, 2020-06-29:
I have the same problem. The reason maybe is the type of CUDA. 
I guess the correspond type of CUDA is 9.0. But i don't want to update the type of CUDA.

And then i have update the type of spconv.
CUDA:10.0
spconv:1.1
pytorch:1.4

fix the problem.
The result of inference is right in Pointpillar,Second,Pv-rcnn.

Gojay001, 2020-07-06:
Hi! I met the same problem, and my enviroment as follows.
```
CUDA:10.0
pytorch:1.1
spconv:1.0
```
Then, I have tried change the version of pytorch, but I got another new problem:
```
ImportError: /home/ease/Documents/OpenLidarPerceptron/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe26detail36_typeMetaDataInstance_preallocated_7E
```
Did you figure out how to solve this? Thx! @	zhanghming @youyousunny

MartinHahner88, 2020-07-11:
maybe [this thread](https://github.com/open-mmlab/OpenPCDet/issues/72#issuecomment-639615724) can help here, too

closed 2020-07-30 by sshaoshuai