open-mmlab/OpenPCDet #71: I got a very, very low accuracy

lan-sky, 2020-05-20:
when i want use pretrained model to test. I got this result:
[image]
I think there are some problem in my dataset. But i can train my model. When i use my model to test. I got the same result:
[image]
I dont know why

GuilinZ, 2020-05-28:
Hi, I have the same problem here, did you figure it out? Thanks! @lan-sky

lan-sky, 2020-06-02:
> Hi, I have the same problem here, did you figure it out? Thanks! @lan-sky

Not yet. I used the spconv1.1 before. It doesnt work when I use spconv1.0
If you solve this problem. Tell me how to solve it. THX @GuilinZ

GuilinZ, 2020-06-02:
> > Hi, I have the same problem here, did you figure it out? Thanks! @lan-sky
> 
> Not yet. I used the spconv1.1 before. It doesnt work when I use spconv1.0
> If you solve this problem. Tell me how to solve it. THX @GuilinZ

Hi I have solved my problem, it is because I didn’t use ‘- - set ‘ setting mentioned in the repo. But while evaluating your own model, you should not use this setting.

lan-sky, 2020-06-02:
> Hi I have solved my problem, it is because I didn’t use ‘- - set ‘ setting mentioned in the repo. But while evaluating your own model, you should not use this setting.
Can you show me the command? I test pointpillar model like this: python3 test.py --cfg_file cfgs/pointpillar.yaml --batch_size 4 --ckpt pointpillar.pth

GuilinZ, 2020-06-02:
> Hi I have solved my problem, it is because I didn’t use ‘- - set ‘ setting mentioned in the repo. But while evaluating your own model, you should not use this setting.
> Can you show me the command? I [...]

I used exactly the same command as in the repo to test the PartA2 model.
python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2.pth \ 
    --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True

GuilinZ, 2020-06-02:
> I used exactly the same command as in the repo to test the PartA2 model.
> python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2.pth \
> --set MODEL.RPN.BACKBONE.NAME UNetV0 [...]

It seems that your problem is not on this setting, since the author said only PartA2 model needs to specify the - - set parameter

lan-sky, 2020-06-02:
> It seems that your problem is not on this setting, since the author said only PartA2 model needs to specify the - - set parameter
Yes, I cant get the result from every model.

sshaoshuai, 2020-06-03:
@lan-sky 
For running inference with the provided `PartA2_car.pth`, please run the following command: 
```
python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2.pth \ 
    --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True
```

But for any other provided models or the models you trained by yourself, you DO NOT need to add the `--set xxx` option.

lan-sky, 2020-06-03:
> @lan-sky
> For running inference with the provided `PartA2_car.pth`, please run the following command:
>
> ```
> python test.py --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2.pth \
>     --set [...]

Thanks for you reply. I know when to add the --set. But I think it is not the problem here. Maybe there are something wrong in my kitti-dataset

Blinjuan, 2020-06-04:
> 
> 
> when i want use pretrained model to test. I got this result:
> [image]
> I think there are some problem in my dataset. But i can train my model. When i use my model to test. I got the same result:
> [image]
> I dont know why

Blinjuan, 2020-06-04:
> 
> 
> when i want use pretrained model to test. I got this result:
> [image]
> I think there are some problem in my dataset. But i can train my model. When i use my model to test. I got the same result:
> [image]
> I dont know why

hi，my problem is different from you，i use command ‘python test.py --cfg_file cfgs/pointpillar.yaml --batch_size 4 --ckpt pointpillar.pth’,and i also get some results
[image]
 I don't know why  i can get so hign accuracy, and i  read the paper” PV-RCNN: Point-Voxel Feature Set Abstraction for 3D Object Detection”，i find the result in paper is lower than me。
is there anything  wrong？

sshaoshuai, 2020-07-17:
This results of our re-implemented PointPillar is corrected.

closed 2020-07-21 by sshaoshuai