open-mmlab/OpenPCDet #72: CUDA error: no kernel image is available for execution on the device

MartinHahner88, 2020-05-20:
Training _PointPillars_ and _SECOND_ works fine, 
but when I try to train _PartA^2Net_, I get this error message after fixing issue [#70](https://github.com/sshaoshuai/PCDet/issues/70):
```
Traceback (most recent call last):
File "train.py", line 215, in <module>
main()
File "train.py", line 210, in main
max_ckpt_save_num=arguments.max_ckpt_save_num)
File "/scratch_net/hox/mhahner/repositories/PCDet/tools/train_utils/train_utils.py", line 80, in train_model
leave_pbar=(cur_epoch + 1 == total_epochs)
File "/scratch_net/hox/mhahner/repositories/PCDet/tools/train_utils/train_utils.py", line 36, in train_one_epoch
loss, tb_dict, disp_dict = model_func(model, batch)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/__init__.py", line 25, in model_func
ret_dict, tb_dict, disp_dict = model(input_dict)
File "/home/mhahner/scratch/apps/anaconda3/envs/spconv/lib/python3.7/site-packages/torch/nn/modules/module.py", line 532, in __call__
result = self.forward(*input, **kwargs)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/detectors/PartA2_net.py", line 112, in forward
batch_size, voxel_centers, coords, rpn_ret_dict, input_dict
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/detectors/PartA2_net.py", line 98, in forward_rcnn
rcnn_ret_dict = self.rcnn_net.forward(rcnn_input_dict)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/rcnn/partA2_rcnn_net.py", line 323, in forward
targets_dict = self.assign_targets(batch_size, rcnn_dict)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/rcnn/partA2_rcnn_net.py", line 27, in assign_targets
targets_dict = proposal_target_layer(rcnn_dict, roi_sampler_cfg=self.rcnn_target_config)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/model_utils/proposal_target_layer.py", line 14, in proposal_target_layer
sample_rois_for_rcnn(rois, gt_boxes, roi_raw_scores, roi_labels, roi_sampler_cfg)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/model_utils/proposal_target_layer.py", line 82, in sample_rois_for_rcnn
cur_gt[:, 0:7], cur_gt_labels)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/models/model_utils/proposal_target_layer.py", line 183, in get_maxiou3d_with_same_class
iou3d = iou3d_nms_utils.boxes_iou3d_gpu(cur_roi, cur_gt) # (M, N)
File "/scratch_net/hox/mhahner/repositories/PCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 47, in boxes_iou3d_gpu
overlaps_h = torch.clamp(min_of_max - max_of_min, min=0)
RuntimeError: CUDA error: no kernel image is available for execution on the device
```

My conda environment looks like this:
```
# packages in environment at /home/mhahner/scratch/apps/anaconda3/envs/spconv:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                      1_llvm    conda-forge
beautifulsoup4            4.9.1                    pypi_0    pypi
bzip2                     1.0.8                h516909a_2    conda-forge
ca-certificates           2020.1.1                      0
cachetools                4.1.0                    pypi_0    pypi
certifi                   2020.4.5.1               py37_0
chardet                   3.0.4                    pypi_0    pypi
cmake                     3.17.0               h28c56e5_0    conda-forge
coloredlogs               14.0                     pypi_0    pypi
cudatoolkit               10.1.243             h6bb024c_0
cudatoolkit-dev           10.1.243             h516909a_3    conda-forge
cudnn                     7.6.5                cuda10.1_0
cycler                    0.10.0                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
easydict                  1.9                      pypi_0    pypi
expat                     2.2.9                he1b5a44_2    conda-forge
google                    2.0.3                    pypi_0    pypi
google-auth               1.15.0                   pypi_0    pypi
google-auth-oauthlib      0.4.1                    pypi_0    pypi
gspread                   3.6.0                    pypi_0    pypi
httplib2                  0.18.0                   pypi_0    pypi
humanfriendly             8.2                      pypi_0    pypi
idna                      2.9                      pypi_0    pypi
imagecodecs               2020.2.18                pypi_0    pypi
imageio                   2.8.0                    pypi_0    pypi
kiwisolver                1.2.0                    pypi_0    pypi
krb5                      1.17.1               h2fd8d38_0    conda-forge
ld_impl_linux-64          2.34                 h53a641e_0    conda-forge
libblas                   3.8.0               16_openblas    conda-forge
libcblas                  3.8.0               16_openblas    conda-forge
libcurl                   7.69.1               hf7181ac_0    conda-forge
libedit                   3.1.20170329      hf8c457e_1001    conda-forge
libffi                    3.2.1             he1b5a44_1007    conda-forge
libgcc-ng                 9.2.0                h24d8f2e_2    conda-forge
libgfortran-ng            7.5.0                hdf63c60_6    conda-forge
liblapack                 3.8.0               16_openblas    conda-forge
libopenblas               0.3.9                h5ec1e0e_0    conda-forge
libssh2                   1.9.0                hab1572f_2    conda-forge
libstdcxx-ng              9.2.0                hdf63c60_2    conda-forge
libuv                     1.34.0               h516909a_0    conda-forge
llvm-openmp               10.0.0               hc9558a2_0    conda-forge
llvmlite                  0.32.1                   pypi_0    pypi
matplotlib                3.2.1                    pypi_0    pypi
mkl                       2020.1                      219    conda-forge
ncurses                   6.1               hf484d3e_1002    conda-forge
networkx                  2.4                      pypi_0    pypi
ninja                     1.10.0               hc9558a2_0    conda-forge
numba                     0.49.1                   pypi_0    pypi
numpy                     1.18.4           py37h8960a57_0    conda-forge
oauth2client              4.1.3                    pypi_0    pypi
oauthlib                  3.1.0                    pypi_0    pypi
openssl                   1.1.1g               h7b6447c_0
pillow                    7.1.2                    pypi_0    pypi
pip                       20.1.1             pyh9f0ad1d_0    conda-forge
protobuf                  3.12.0                   pypi_0    pypi
pyasn1                    0.4.8                    pypi_0    pypi
pyasn1-modules            0.2.8                    pypi_0    pypi
pyparsing                 2.4.7                    pypi_0    pypi
python                    3.7.6           h8356626_5_cpython    conda-forge
python-dateutil           2.8.1                    pypi_0    pypi
python_abi                3.7                     1_cp37m    conda-forge
pytorch                   1.4.0           py3.7_cuda10.1.243_cudnn7.6.3_0    pytorch
pywavelets                1.1.1                    pypi_0    pypi
pyyaml                    5.3.1                    pypi_0    pypi
readline                  8.0                  hf8c457e_0    conda-forge
requests                  2.23.0                   pypi_0    pypi
requests-oauthlib         1.3.0                    pypi_0    pypi
rhash                     1.3.6             h14c3975_1001    conda-forge
rsa                       4.0                      pypi_0    pypi
scikit-image              0.17.2                   pypi_0    pypi
scipy                     1.4.1                    pypi_0    pypi
setuptools                46.4.0           py37hc8dfbb8_0    conda-forge
six                       1.14.0                   pypi_0    pypi
soupsieve                 2.0.1                    pypi_0    pypi
spconv                    1.0                      pypi_0    pypi
sqlite                    3.30.1               hcee41ef_0    conda-forge
tensorboardx              2.0                      pypi_0    pypi
tifffile                  2020.5.11                pypi_0    pypi
tk                        8.6.10               hed695b0_0    conda-forge
tqdm                      4.46.0                   pypi_0    pypi
urllib3                   1.25.9                   pypi_0    pypi
wheel                     0.34.2                     py_1    conda-forge
xz                        5.2.5                h516909a_0    conda-forge
zlib                      1.2.11            h516909a_1006    conda-forge
```

codyreading, 2020-05-23:
@MartinHahner88 Have you tried using Docker? That should create a consistent environment.

MartinHahner88, 2020-05-25:
I wish I could use Docker, but in our lab, we, unfortunately, have no possibility to use Docker. :(

Via Google, I think I found two similar issues, one with [pytorch](https://github.com/pytorch/pytorch/issues/31285) and one with [detectron2](https://github.com/facebookresearch/detectron2/issues/235), 
but I could not get a solution out of these threads either.

@codyreading: Can you make any sense of the additional information there?

codyreading, 2020-05-25:
Hi @MartinHahner88,

Sorry, I am not too familar with original code, especially PartA2. I'm not entirely sure why this occuring.

sshaoshuai, 2020-06-02:
Hi MartinHahner, 
I didn't test it on PyTorch 1.4 since I generally use PyTorch 1.1 due to the problem of low version Nvidia-driver. Have you tried to run it with PyTorch 1.1? 
Also, what's your NVIDIA driver and GPU?

MartinHahner88, 2020-06-02:
@sshaoshuai 
I am dealing with Driver Version: 390.116 and a GTX 1060 on my work desktop 
(where your repository does not necessarily need to run), 
but more importantly, I can not make it run on our GPU cluster with **Driver Version: 418.74** 
and a variety of those GPUs
- GTX 1080 Ti
- GTX TITAN X
- Titan Xp
- Titan X
- RTX 2080 Ti

sshaoshuai, 2020-06-02:
@MartinHahner88 
Your work desktop is the same with my desktop and I could install it successfully. 
`PCDet` has also been tested on `GTX 1080 Ti, Titan Xp, Titan X` but with PyTorch 1.1 and older nvidia driver and CUDA 9.0. 

The error log of your cmake seems strange. Have you tried to clean the folder (rm `build` and run `python setup.py clean`) and re-build the `spconv 1.0`? 
Make sure to use `gcc 5.4` and `cmake 3.13+`.

MartinHahner88, 2020-06-02:
@sshaoshuai 
Thanks for the hint about `gcc`.
> Make sure to use `gcc 5.4` and `cmake 3.13`

```
$ gcc --version
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
```

Our gcc version is newer, so I will try to see if I can downgrade to or separately install `gcc 5.4` 
on our cluster and report back if that finally resolves everything.

sshaoshuai, 2020-06-03:
@MartinHahner88 
Please also make sure to rm `build` and run `python setup.py clean` before the next building to reduce the potential risk from building history.

MartinHahner88, 2020-06-03:
Yes, I made sure to run `rm - rf build` and `python setup.py clean` before every new build attempt.

MartinHahner88, 2020-06-04:
I installed gcc v.5.4.0 now.
Still same behavior though when I try to build spconv v1.0 (commit 8da6f96).

```
$ conda list
# packages in environment at /home/mhahner/scratch/apps/anaconda3/envs/PCDet:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main
binutils_impl_linux-64    2.28.1               had2808c_3
binutils_linux-64         5.4.0               had2808c_24
blas                      1.0                         mkl
bzip2                     1.0.8                h7b6447c_0
ca-certificates           2020.1.1                      0
certifi                   2020.4.5.1               py36_0
cffi                      1.14.0           py36he30daa8_1
cmake                     3.14.0               h52cb24c_0
cudatoolkit               9.0                  h13b8566_0
cudnn                     7.6.5                 cuda9.0_0
expat                     2.2.6                he6710b0_0
freetype                  2.9.1                h8a8886c_1
gcc_impl_linux-64         5.4.0                habb00fd_3
gcc_linux-64              5.4.0               h98af8de_24
gxx_impl_linux-64         5.4.0                hdf63c60_3
gxx_linux-64              5.4.0               h98af8de_24
intel-openmp              2020.1                      217
jpeg                      9b                   h024ee3a_2
krb5                      1.17.1               h173b8e3_0
ld_impl_linux-64          2.33.1               h53a641e_7
libcurl                   7.69.1               h20c2e04_0
libedit                   3.1.20181209         hc058e9b_0
libffi                    3.3                  he6710b0_1
libgcc-ng                 9.1.0                hdf63c60_0
libgfortran-ng            7.3.0                hdf63c60_0
libpng                    1.6.37               hbc83047_0
libssh2                   1.9.0                h1ba5d50_1
libstdcxx-ng              9.1.0                hdf63c60_0
libtiff                   4.1.0                h2733197_0
mkl                       2020.1                      217
mkl-service               2.3.0            py36he904b0f_0
mkl_fft                   1.0.15           py36ha843d7b_0
mkl_random                1.1.1            py36h0573a6f_0
ncurses                   6.2                  he6710b0_1
ninja                     1.9.0            py36hfd86e86_0
numpy                     1.18.1           py36h4f9e942_0
numpy-base                1.18.1           py36hde5b4d6_1
olefile                   0.46                     py36_0
openssl                   1.1.1g               h7b6447c_0
pillow                    7.1.2            py36hb39fc2d_0
pip                       20.0.2                   py36_3
pycparser                 2.20                       py_0
python                    3.6.10               h7579374_2
pytorch                   1.1.0           py3.6_cuda9.0.176_cudnn7.5.1_0    pytorch
readline                  8.0                  h7b6447c_0
rhash                     1.3.8                h1ba5d50_0
setuptools                46.4.0                   py36_0
six                       1.14.0                   py36_0
sqlite                    3.31.1               h62c20be_1
tk                        8.6.8                hbc83047_0
torchvision               0.3.0           py36_cu9.0.176_1    pytorch
wheel                     0.34.2                   py36_0
xz                        5.2.5                h7b6447c_0
zlib                      1.2.11               h7b6447c_3
zstd                      1.3.7                h0b5b093_0
```

```
(PCDet) mhahner@hox:/scratch_net/hox/mhahner/repositories/spconv_8da6f96 ((HEAD detached from 8da6f96)) 
$ CUDA_ROOT=/scratch_net/hox/mhahner/apps/cuda-9.1 python setup.py bdist_wheel
running bdist_wheel
[... 576 lines cut ...]
    dist.run_commands()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 223, in run
    self.run_command('build')
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 39, in run
    self.build_extension(ext)
  File "setup.py", line 87, in build_extension
    subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
```

sshaoshuai, 2020-06-04:
@MartinHahner88 
You need to install boost firstly, please try the following step: 
1. Install the boost via anaconda
```
conda install libboost
```
2. Include the boost directory by adding the following line to CMakeList.txt, just after the `project xxx` is ok: 
```
include_directories(/PATH_TO_YOUR_ANACONDA/anaconda3/envs/PCDet/include)
```
3. Try to rebuild your spconv

MartinHahner88, 2020-06-04:
@sshaoshuai: Thanks for still helping me out here!

I successfully installed `libboost`via `conda install libboost`, 
but maybe I did not get your 2. point, because the build still crashed.

Did you mean it like that (see below) in `CMaleList.txt`?

cmake_minimum_required(VERSION 3.13 FATAL_ERROR)
project(SparseConv LANGUAGES CXX CUDA VERSION 1.0)
**include_directories(/home/mhahner/scratch/apps/anaconda3/envs/PCDet/include)**

option(SPCONV_BuildTests "Build the unit tests when BUILD_TESTING is enabled." ON)
set(CMAKE_CXX_EXTENSIONS OFF) # avoid gnu++11 be added to CXX flags

sshaoshuai, 2020-06-04:
Yes, so what's your error now? Could you post the build log like before?

MartinHahner88, 2020-06-04:
```
(PCDet) mhahner@hox:/scratch_net/hox/mhahner/repositories/spconv_8da6f96 ((HEAD detached from 8da6f96))
$ CUDA_ROOT=/scratch_net/hox/mhahner/apps/cuda-9.1 python setup.py bdist_wheel

running bdist_wheel
running build
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/spconv
copying spconv/functional.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/ops.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/pool.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/modules.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/conv.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/test_utils.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/__init__.py -> build/lib.linux-x86_64-3.6/spconv
creating build/lib.linux-x86_64-3.6/spconv/utils
copying spconv/utils/__init__.py -> build/lib.linux-x86_64-3.6/spconv/utils
running build_ext
/scratch_net/hox/mhahner/repositories/spconv_8da6f96/build/lib.linux-x86_64-3.6
Release
-- The CXX compiler identification is GNU 5.4.0
-- The CUDA compiler identification is NVIDIA 9.1.85
-- Check for working CXX compiler: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/bin/x86_64-conda_cos6-linux-gnu-c++
-- Check for working CXX compiler: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/bin/x86_64-conda_cos6-linux-gnu-c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working CUDA compiler: /scratch_net/hox/mhahner/apps/cuda-9.1/bin/nvcc
-- Check for working CUDA compiler: /scratch_net/hox/mhahner/apps/cuda-9.1/bin/nvcc -- works
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /scratch_net/hox/mhahner/apps/cuda-9.1 (found version "9.1") 
-- Caffe2: CUDA detected: 9.1
-- Caffe2: CUDA nvcc is: /scratch_net/hox/mhahner/apps/cuda-9.1/bin/nvcc
-- Caffe2: CUDA toolkit directory: /scratch_net/hox/mhahner/apps/cuda-9.1
-- Caffe2: Header version is: 9.1
-- Found CUDNN: /scratch_net/hox/mhahner/apps/anaconda3/envs/PCDet/include  
-- Found cuDNN: v7.6.5  (include: /scratch_net/hox/mhahner/apps/anaconda3/envs/PCDet/include, library: /scratch_net/hox/mhahner/apps/anaconda3/envs/PCDet/lib/libcudnn.so)
-- Autodetected CUDA architecture(s):  6.1
-- Added CUDA NVCC flags for: -gencode;arch=compute_61,code=sm_61
-- Found torch: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/lib/libtorch.so  
-- Found PythonInterp: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/bin/python3.6 (found suitable version "3.6.10", minimum required is "3.6") 
-- Found PythonLibs: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/libpython3.6m.so
-- Performing Test HAS_CPP14_FLAG
-- Performing Test HAS_CPP14_FLAG - Success
-- pybind11 v2.3.dev0
-- Configuring done
-- Generating done
-- Build files have been written to: /scratch_net/hox/mhahner/repositories/spconv_8da6f96/build/temp.linux-x86_64-3.6
Scanning dependencies of target spconv_nms
Scanning dependencies of target spconv
[  7%] Building CUDA object src/utils/CMakeFiles/spconv_nms.dir/nms.cu.o
[ 21%] Building CXX object src/spconv/CMakeFiles/spconv.dir/indice.cc.o
[ 28%] Building CUDA object src/spconv/CMakeFiles/spconv.dir/indice.cu.o
[ 21%] Building CXX object src/spconv/CMakeFiles/spconv.dir/all.cc.o
[ 35%] Linking CUDA device code CMakeFiles/spconv_nms.dir/cmake_device_link.o
[ 42%] Linking CUDA shared library ../../../lib.linux-x86_64-3.6/spconv/libspconv_nms.so
[ 42%] Built target spconv_nms
[... 506 lines cut ...]
    dist.run_commands()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 223, in run
    self.run_command('build')
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 39, in run
    self.build_extension(ext)
  File "setup.py", line 87, in build_extension
    subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
```

sshaoshuai, 2020-06-04:
@MartinHahner88 
Ok, the boost error is gone now. 

I notice another error in your build log: 
```
torch_utils.h: In function 'tv::TensorView<T> tv::torch2tv(const at::Tensor&)':
```
The reason may be the un-matched version of CUDA and pytorch. 
Your pytorch version is  `py3.6_cuda9.0.176_cudnn7.5.1_0`, but when  you build the spconv, your CUDA version is `CUDA 9.1`.
Could you use CUDA 9.0 to build spconv?

MartinHahner88, 2020-06-04:
conda environment unchanged:

```
(PCDet) mhahner@biwirender04:/scratch_net/hox/mhahner/repositories/spconv_8da6f96 ((HEAD detached from 8da6f96))
$ CUDA_ROOT=/scratch_net/hox/mhahner/apps/cuda-9.0 python setup.py bdist_wheel

running bdist_wheel
running build
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/spconv
copying spconv/functional.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/ops.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/pool.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/modules.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/conv.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/test_utils.py -> build/lib.linux-x86_64-3.6/spconv
copying spconv/__init__.py -> build/lib.linux-x86_64-3.6/spconv
creating build/lib.linux-x86_64-3.6/spconv/utils
copying spconv/utils/__init__.py -> build/lib.linux-x86_64-3.6/spconv/utils
running build_ext
/scratch_net/hox/mhahner/repositories/spconv_8da6f96/build/lib.linux-x86_64-3.6
Release
-- The CXX compiler identification is GNU 5.4.0
-- The CUDA compiler identification is NVIDIA 9.0.176
-- Check for working CXX compiler: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/bin/x86_64-conda_cos6-linux-gnu-c++
-- Check for working CXX compiler: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/bin/x86_64-conda_cos6-linux-gnu-c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working CUDA compiler: /scratch_net/hox/mhahner/apps/cuda-9.0/bin/nvcc
-- Check for working CUDA compiler: /scratch_net/hox/mhahner/apps/cuda-9.0/bin/nvcc -- works
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /scratch_net/hox/mhahner/apps/cuda-9.0 (found version "9.0") 
-- Caffe2: CUDA detected: 9.0
-- Caffe2: CUDA nvcc is: /scratch_net/hox/mhahner/apps/cuda-9.0/bin/nvcc
-- Caffe2: CUDA toolkit directory: /scratch_net/hox/mhahner/apps/cuda-9.0
-- Caffe2: Header version is: 9.0
-- Found CUDNN: /scratch_net/hox/mhahner/apps/cuda-9.0/include  
-- Found cuDNN: v7.1.3  (include: /scratch_net/hox/mhahner/apps/cuda-9.0/include, library: /scratch_net/hox/mhahner/apps/cuda-9.0/lib64/libcudnn.so)
-- Autodetected CUDA architecture(s):  3.5
-- Added CUDA NVCC flags for: -gencode;arch=compute_35,code=sm_35
-- Found torch: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/lib/libtorch.so  
-- Found PythonInterp: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/bin/python3.6 (found suitable version "3.6.10", minimum required is "3.6") 
-- Found PythonLibs: /home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/libpython3.6m.so
-- Performing Test HAS_CPP14_FLAG
-- Performing Test HAS_CPP14_FLAG - Success
-- pybind11 v2.3.dev0
-- Configuring done
CMake Warning at src/spconv/CMakeLists.txt:1 (add_library):
  Cannot generate a safe runtime search path for target spconv because there
  is a cycle in the constraint graph:

    dir 0 is [/scratch_net/hox/mhahner/apps/cuda-9.0/lib64]
      dir 3 must precede it due to runtime library [libnvrtc.so.9.0]
    dir 1 is [/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/lib]
    dir 2 is [/scratch_net/hox/mhahner/apps/cuda-9.0/lib64/stubs]
    dir 3 is [/scratch_net/hox/mhahner/apps/anaconda3/envs/PCDet/lib]
[... 520 lines cut ...]
    dist.run_commands()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 223, in run
    self.run_command('build')
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 39, in run
    self.build_extension(ext)
  File "setup.py", line 87, in build_extension
    subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
  File "/home/mhahner/scratch/apps/anaconda3/envs/PCDet/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
```

sshaoshuai, 2020-06-04:
@MartinHahner88 
Sorry, I have no idea about it....
But I saw others solved this problem by re-installing pytorch and torchvision, maybe you could have a try.

MartinHahner88, 2020-06-04:
@sshaoshuai: Thanks for your efforts, the hint about `libboost`  was still useful. 

Trying to use a CUDA 9.2 environment unfortunately also did not succeed.

```
(cuda9-2) mhahner@hox:/scratch_net/hox/mhahner/repositories/spconv_8da6f96 ((HEAD detached from 8da6f96))
$ conda list
# packages in environment at /home/mhahner/scratch/apps/anaconda3/envs/cuda9-2:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main
binutils_impl_linux-64    2.28.1               had2808c_3
binutils_linux-64         5.4.0               had2808c_24
blas                      1.0                         mkl
boost                     1.67.0                   py37_4
bzip2                     1.0.8                h7b6447c_0
ca-certificates           2020.1.1                      0
certifi                   2020.4.5.1               py37_0
cffi                      1.14.0           py37h2e261b9_0
chardet                   3.0.4                 py37_1003
cmake                     3.17.0               h28c56e5_0    conda-forge
conda                     4.8.3            py37hc8dfbb8_1    conda-forge
conda-package-handling    1.6.1            py37h7b6447c_0
cryptography              2.9.2            py37h1ba5d50_0
cudatoolkit               9.2                           0
cudatoolkit-dev           9.2                      py37_1    conda-forge
cudnn                     7.6.5                 cuda9.2_0
expat                     2.2.9                he1b5a44_2    conda-forge
freetype                  2.9.1                h8a8886c_1
gcc_impl_linux-64         5.4.0                habb00fd_3
gcc_linux-64              5.4.0               h98af8de_24
gxx_impl_linux-64         5.4.0                hdf63c60_3
gxx_linux-64              5.4.0               h98af8de_24
icu                       58.2                 he6710b0_3
idna                      2.9                        py_1
intel-openmp              2020.1                      217
jpeg                      9b                   h024ee3a_2
krb5                      1.17.1               h173b8e3_0
ld_impl_linux-64          2.33.1               h53a641e_7
libboost                  1.67.0               h46d08c1_4
libcurl                   7.69.1               h20c2e04_0
libedit                   3.1.20181209         hc058e9b_0
libffi                    3.2.1                hd88cf55_4
libgcc-ng                 9.1.0                hdf63c60_0
libgfortran-ng            7.3.0                hdf63c60_0
libpng                    1.6.37               hbc83047_0
libssh2                   1.9.0                h1ba5d50_1
libstdcxx-ng              9.1.0                hdf63c60_0
libtiff                   4.1.0                h2733197_1
libuv                     1.26.0               h7b6447c_0
lz4-c                     1.9.2                he6710b0_0
mkl                       2020.1                      217
mkl-service               2.3.0            py37he904b0f_0
mkl_fft                   1.0.15           py37ha843d7b_0
mkl_random                1.1.1            py37h0573a6f_0
ncurses                   6.1                  he6710b0_1
ninja                     1.9.0            py37hfd86e86_0
numpy                     1.18.1           py37h4f9e942_0
numpy-base                1.18.1           py37hde5b4d6_1
olefile                   0.46                     py37_0
openssl                   1.1.1g               h7b6447c_0
pillow                    7.1.2            py37hb39fc2d_0
pip                       20.0.2                   py37_3
py-boost                  1.67.0           py37h04863e7_4
pycosat                   0.6.3            py37h7b6447c_0
pycparser                 2.20                       py_0
pyopenssl                 19.1.0                   py37_0
pysocks                   1.7.1                    py37_0
python                    3.7.6           cpython_h8356626_6    conda-forge
python_abi                3.7                     1_cp37m    conda-forge
[... 420 lines cut ...]
    dist.run_commands()
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 223, in run
    self.run_command('build')
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 39, in run
    self.build_extension(ext)
  File "setup.py", line 87, in build_extension
    subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
  File "/home/mhahner/scratch/apps/anaconda3/envs/cuda9-2/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
```

sshaoshuai, 2020-06-04:
I have no idea about that ...
Maybe you could create a new environment and  try to install pytorch with pip, like `pip install torch==1.1`, and I generally use cuda 9.0 for pytorch 1.1.

MartinHahner88, 2020-06-05:
Finally SOLVED!

The final issue was, that **we tried to build spconv on a _Tesla K40_** GPU, 
which has a _Kepler_ architecture and is seemingly too old to build _spconv_.

_The reason why we always tried to build spconv on a Tesla K40 is that in our lab, it is hard to get a GPU on our GPU cluster interactively. (Usually, you can only submit jobs via qsub/SLURM.)
So we always tried to build spconv either on our local Linux clients or on old nodes which only have Tesla K40s. Then, out of despair, we tried to build spconv on a Titan X (Pascal) and we finally found a combination of requirements that worked._

So here is our solution:

```
conda create --name PCDet python=3.6 pytorch=1.1 cudatoolkit=9.2 cudatoolkit-dev=9.2 \
cmake --channel pytorch --channel=conda-forge
conda activate PCDet
conda install cudnn
conda install boost
git clone https://github.com/traveller59/spconv spconv_8da6f96 --recursive
cd spconv_8da6f96
git checkout 8da6f967fb9a054d8870c3515b1b44eca2103634
```

> If necessary: 
> Download and extract 
> [0001-Allow-to-specifiy-CUDA_ROOT-directory-and-pick-corre.patch.zip](https://github.com/sshaoshuai/PCDet/files/4737235/0001-Allow-to-specifiy-CUDA_ROOT-directory-and-pick-corre.patch.zip)
> and patch _spconv_ via:
> ```git am <PATH_TO_EXTRACTED_FILE>/0001-Allow-to-specifiy-CUDA_ROOT-directory-and-pick-corre.patch```

```
CUDA_ROOT=<PATH_TO_YOUR_CONDA_INSTALLATION>/conda_envs/PCDet python setup.py bdist_wheel
cd dist/
pip install *
```
Test _spconv_ via:
`python -c 'import spconv'`
(should just return and not raise any errors) 

```
cd ../..
git clone https://github.com/sshaoshuai/PCDet.git
cd PCDet/
pip install -r requirements.txt
CUDA_ROOT=<PATH_TO_YOUR_CONDA_INSTALLATION>/conda_envs/PCDet python setup.py develop
```

Done!

I hope these instructions help someone else who struggles to build _spconv_ as well.

closed 2020-06-05 by MartinHahner88

sshaoshuai, 2020-06-05:
@MartinHahner88 
Great! 
Thank you for the detailed instruction!

MartinHahner88, 2020-07-11:
@sshaoshuai: maybe add [these instructions](https://github.com/open-mmlab/OpenPCDet/issues/72#issuecomment-639615724) to your [INSTALL.md](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md)

it seems to be the way to go for many people to get the repository running 🏃

KangChou, 2021-06-24:
` CUDA_ROOT=/usr/local/cuda-11.0 python setup.py develop
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
/opt/conda/lib/python3.6/site-packages/torch/utils/cpp_extension.py:339: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
gcc -pthread -B /opt/conda/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/opt/conda/lib/python3.6/site-packages/torch/include -I/opt/conda/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/lib/python3.6/site-packages/torch/include/TH -I/opt/conda/lib/python3.6/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/include/python3.6m -c pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o build/temp.linux-x86_64-3.6/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
In file included from /opt/conda/lib/python3.6/site-packages/torch/include/ATen/Parallel.h:149:0,
                 from /opt/conda/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/utils.h:3,
                 from /opt/conda/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:5,
                 from /opt/conda/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/nn.h:3,
                 from /opt/conda/lib/python3.6/site-packages/torch/include/torch/csrc/api/include/torch/all.h:12,
                 from /opt/conda/lib/python3.6/site-packages/torch/include/torch/extension.h:4,
                 from pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:10:
/opt/conda/lib/python3.6/site-packages/torch/include/ATen/ParallelOpenMP.h:84:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
 #pragma omp parallel for if ((end - begin) >= grain_size)
 ^
pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:12:18: fatal error: cuda.h: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit status 1
`

KangChou, 2021-06-24:
[image]