open-mmlab/OpenPCDet #73: ImportError: from . import roiaware_pool3d_cuda

Michael-Shaw, 2020-05-25:
### When I run the code:
python kitti_dataset.py create_kitti_infos
### The error occurred:
Traceback (most recent call last):
  File "kitti_dataset.py", line 11, in <module>
    from pcdet.utils import box_utils, object3d_utils, calibration, common_utils
  File "/home/jhvision5/Michael/pyPrj/PCDet/pcdet/utils/box_utils.py", line 5, in <module>
    from ..ops.roiaware_pool3d import roiaware_pool3d_utils
  File "/home/jhvision5/Michael/pyPrj/PCDet/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 4, in <module>
    from . import roiaware_pool3d_cuda
ImportError: libcudart.so.8.0: cannot open shared object file: No such file or directory
### PCDet installation seems to be no problem:
running develop
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
copying build/lib.linux-x86_64-3.7/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-37m-x86_64-linux-gnu.so -> pcdet/ops/iou3d_nms
copying build/lib.linux-x86_64-3.7/pcdet/ops/roiaware_pool3d/roiaware_pool3d_cuda.cpython-37m-x86_64-linux-gnu.so -> pcdet/ops/roiaware_pool3d
Creating /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages/pcdet.egg-link (link to .)
pcdet 0.1.0+2244be4 is already the active version in easy-install.pth

Installed /home/jhvision5/Michael/pyPrj/PCDet
Processing dependencies for pcdet==0.1.0+2244be4
Searching for PyYAML==5.1.2
Best match: PyYAML 5.1.2
Adding PyYAML 5.1.2 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for easydict==1.9
Best match: easydict 1.9
Adding easydict 1.9 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for tensorboardX==1.9
Best match: tensorboardX 1.9
Adding tensorboardX 1.9 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for numba==0.46.0
Best match: numba 0.46.0
Adding numba 0.46.0 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for spconv==1.0
Best match: spconv 1.0
Adding spconv 1.0 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for torch==1.1.0
Best match: torch 1.1.0
Adding torch 1.1.0 to easy-install.pth file
Installing convert-caffe2-to-onnx script to /home/jhvision5/anaconda3/envs/michael/bin
Installing convert-onnx-to-caffe2 script to /home/jhvision5/anaconda3/envs/michael/bin

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for numpy==1.17.3
Best match: numpy 1.17.3
Adding numpy 1.17.3 to easy-install.pth file
Installing f2py script to /home/jhvision5/anaconda3/envs/michael/bin
Installing f2py3 script to /home/jhvision5/anaconda3/envs/michael/bin
Installing f2py3.7 script to /home/jhvision5/anaconda3/envs/michael/bin

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for protobuf==3.11.1
Best match: protobuf 3.11.1
Adding protobuf 3.11.1 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for six==1.13.0
Best match: six 1.13.0
Adding six 1.13.0 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for llvmlite==0.30.0
Best match: llvmlite 0.30.0
Adding llvmlite 0.30.0 to easy-install.pth file

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Searching for setuptools==39.1.0
Best match: setuptools 39.1.0
Adding setuptools 39.1.0 to easy-install.pth file
Installing easy_install script to /home/jhvision5/anaconda3/envs/michael/bin
Installing easy_install-3.6 script to /home/jhvision5/anaconda3/envs/michael/bin

Using /home/jhvision5/anaconda3/envs/michael/lib/python3.7/site-packages
Finished processing dependencies for pcdet==0.1.0+2244be4
### My cuda version is 9.0, why is there an "ImportError: libcudart.so.8.0" error: 
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Sep__1_21:08:03_CDT_2017
Cuda compilation tools, release 9.0, V9.0.176
How can I solve this problem, can anyone help me answer it, thank you!

sventeufel, 2020-05-26:
I had a similar error, upgrading from spconv 1.0 to spconv 1.1 solved it for me

sshaoshuai, 2020-06-02:
Please make sure to buid pcdet with CUDA 9.0. 
You could remove the `build` folder and try to rebuild pcdet with CUDA 9.0

closed 2020-07-17 by sshaoshuai

A-Janj, 2023-02-21:
> Please make sure to buid pcdet with CUDA 9.0. You could remove the `build` folder and try to rebuild pcdet with CUDA 9.0

Can this repo not be used with CUDA 11.6?

continue-up, 2024-04-02:
> > Please make sure to buid pcdet with CUDA 9.0. You could remove the `build` folder and try to rebuild pcdet with CUDA 9.0
> 
> Can this repo not be used with CUDA 11.6?

I have the same question, could this repo be used with CUDA 11.6?