open-mmlab/OpenPCDet #74: I get high accuracy during evaluation, but many results don't look good in kitti_object_vis

tleyden, 2020-05-26:
System info

| Component  | Version |
| ------------- | ------------- |
| OS  | Ubuntu 18.04  |
| Nvidia Driver  | 440.82  |
| Python | 3.6.9 | 
| Cuda  | 10.2 |
| Torch  | 1.4  |
| PCDet | 2244be47d3721ba95cf812c8646ffa305b153db4 Sun Apr 26 23:42:49 2020 | 
| spconv  | v1.0 (commit 8da6f96)  |

I'm downloading the [Part-A^2 model-209M](https://drive.google.com/open?id=1D-lxyPww80H-zEdheaDTO6BfxCFiXOEo) model and evaluating with:

```
python test.py --save_to_file --cfg_file cfgs/PartA2_car.yaml 
  --batch_size 4 
  --ckpt PartA2_car.pth 
  --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True
```

with output:

```
2020-05-25 21:14:37,876   INFO  *************** Performance of EPOCH 2 *****************
2020-05-25 21:14:37,877   INFO  Generate label finished(sec_per_example: 0.1206 second).
2020-05-25 21:14:37,877   INFO  recall_roi_0.5: 0.964964
2020-05-25 21:14:37,877   INFO  recall_rcnn_0.5: 0.965103
2020-05-25 21:14:37,877   INFO  recall_roi_0.7: 0.815572
2020-05-25 21:14:37,877   INFO  recall_rcnn_0.7: 0.848523
2020-05-25 21:14:37,878   INFO  Average predicted number of objects(3769 samples): 5.386
```

<details>
<summary>Full command output</summary>

```
$ python test.py --save_to_file --cfg_file cfgs/PartA2_car.yaml --batch_size 4 --ckpt PartA2_car.pth --set MODEL.RPN.BACKBONE.NAME UNetV0 MODEL.RPN.RPN_HEAD.ARGS use_binary_dir_classifier:True
/home/tleyden/Development/PCDet/pcdet/datasets/data_augmentation/augmentation_utils.py:11: NumbaDeprecationWarning: An import was requested from a module that has moved location.
Import requested from: 'numba.errors', please update to use 'numba.core.errors' or pin to Numba version 0.48.0. This alias will not be present in Numba version 0.50.0.
  from numba.errors import NumbaPerformanceWarning
/home/tleyden/Development/PCDet/pcdet/datasets/data_augmentation/augmentation_utils.py:11: NumbaDeprecationWarning: An import was requested from a module that has moved location.
Import of 'NumbaPerformanceWarning' requested from: 'numba.errors', please update to use 'numba.core.errors' or pin to Numba version 0.48.0. This alias will not be present in Numba version 0.50.0.
  from numba.errors import NumbaPerformanceWarning
2020-05-25 21:07:00,659   INFO  **********************Start logging**********************
2020-05-25 21:07:00,659   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-05-25 21:07:00,659   INFO  cfg_file         cfgs/PartA2_car.yaml
2020-05-25 21:07:00,659   INFO  data_dir         None
2020-05-25 21:07:00,659   INFO  batch_size       4
2020-05-25 21:07:00,659   INFO  epochs           80
2020-05-25 21:07:00,659   INFO  workers          4
2020-05-25 21:07:00,659   INFO  extra_tag        default
2020-05-25 21:07:00,659   INFO  ckpt             PartA2_car.pth
2020-05-25 21:07:00,659   INFO  mgpus            False
2020-05-25 21:07:00,659   INFO  launcher         none
2020-05-25 21:07:00,659   INFO  tcp_port         18888
2020-05-25 21:07:00,659   INFO  local_rank       0
2020-05-25 21:07:00,659   INFO  set_cfgs         ['MODEL.RPN.BACKBONE.NAME', 'UNetV0', 'MODEL.RPN.RPN_HEAD.ARGS', 'use_binary_dir_classifier:True']
2020-05-25 21:07:00,660   INFO  max_waiting_mins 30
2020-05-25 21:07:00,660   INFO  start_epoch      0
2020-05-25 21:07:00,660   INFO  eval_tag         default
2020-05-25 21:07:00,660   INFO  eval_all         False
2020-05-25 21:07:00,660   INFO  ckpt_dir         None
2020-05-25 21:07:00,660   INFO  save_to_file     True
2020-05-25 21:07:00,660   INFO  cfg.ROOT_DIR: /home/tleyden/Development/PCDet
2020-05-25 21:07:00,660   INFO  cfg.LOCAL_RANK: 0
2020-05-25 21:07:00,660   INFO  cfg.CLASS_NAMES: ['Car']
2020-05-25 21:07:00,660   INFO
cfg.DATA_CONFIG = edict()
2020-05-25 21:07:00,660   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
[... 218 lines cut ...]
When I copy the files in `PCDet/output/PartA2_car/default/eval/epoch_2/val/default` to `../data/kitti/testing/pred` and run `kitti-vis` with:

```
python kitti_object.py -d ~/Development/PCDet/data/kitti/ 
  --show_lidar_with_depth 
  --img_fov 
  --const_box 
  --vis 
  --show_image_with_boxes 
  --pred  
  --ind 21 
  --split testing
```

it's giving this result:

[image]

which is not detecting the car, but is detecting many other bounding boxes for things that don't appear in the pointcloud data.   

Some examples are a bit better, for example if I use `--ind 4` it seems to at least detect the car to the left:

[image]

But overall the results seem far less than 78% accurate after checking about 20 results.  Anything obvious I'm doing wrong with either the evaluation stage or how I'm calling kitti-vis?

tleyden, 2020-05-26:
I discovered the issue -- I was using assuming that `test.py` was happening on the kitti data `TESTING` set, whereas it was actually happening over the validation subset of the `TRAINING` set.  

After copying  `PCDet/output/pointpillar/default/eval/epoch_no_number/val/default/final_result/data/`  to `../data/kitti/training/pred` (ignore the fact that it's using pointpillar now, probably not consequential), I'm able to run kitti-vis with:

```
python kitti_object.py -d ~/Development/PCDet/data/kitti/ 
  --show_lidar_with_depth 
  --img_fov 
  --const_box 
  --vis 
  --show_image_with_boxes 
  --pred  
  --ind 4 
```

and it gives accurate results now (the prediction bounding box is in red, wheras as the ground truth label is in green):

[image]

The exact `test.py` command I used to generate predictions for pointpillar was:

```
python test.py --save_to_file --cfg_file cfgs/pointpillar_high_thresh.yaml --batch_size 4 --ckpt pointpillar.pth
```

where `pointpillar_high_thresh.yaml` is a copy of `pointpillar.yaml` with `threshold=0.4`.

<details>
<summary>Command output (expand)</summary>

```
python test.py --save_to_file --cfg_file cfgs/pointpillar_high_thresh.yaml --batch_size 4 --ckpt pointpillar.pth

/home/tleyden/Development/PCDet/pcdet/datasets/data_augmentation/augmentation_utils.py:11: NumbaDeprecationWarning: An import was requested from a module that has moved location.
Import requested from: 'numba.errors', please update to use 'numba.core.errors' or pin to Numba version 0.48.0. This alias will not be present in Numba version 0.50.0.
  from numba.errors import NumbaPerformanceWarning
/home/tleyden/Development/PCDet/pcdet/datasets/data_augmentation/augmentation_utils.py:11: NumbaDeprecationWarning: An import was requested from a module that has moved location.
Import of 'NumbaPerformanceWarning' requested from: 'numba.errors', please update to use 'numba.core.errors' or pin to Numba version 0.48.0. This alias will not be present in Numba version 0.50.0.
  from numba.errors import NumbaPerformanceWarning
2020-05-25 23:00:44,154   INFO  **********************Start logging**********************
2020-05-25 23:00:44,154   INFO  CUDA_VISIBLE_DEVICES=ALL
2020-05-25 23:00:44,154   INFO  cfg_file         cfgs/pointpillar_high_thresh.yaml
2020-05-25 23:00:44,154   INFO  data_dir         None
2020-05-25 23:00:44,154   INFO  batch_size       4
2020-05-25 23:00:44,155   INFO  epochs           80
2020-05-25 23:00:44,155   INFO  workers          4
2020-05-25 23:00:44,155   INFO  extra_tag        default
2020-05-25 23:00:44,155   INFO  ckpt             pointpillar.pth
2020-05-25 23:00:44,155   INFO  mgpus            False
2020-05-25 23:00:44,155   INFO  launcher         none
2020-05-25 23:00:44,155   INFO  tcp_port         18888
2020-05-25 23:00:44,155   INFO  local_rank       0
2020-05-25 23:00:44,155   INFO  set_cfgs         None
2020-05-25 23:00:44,155   INFO  max_waiting_mins 30
2020-05-25 23:00:44,155   INFO  start_epoch      0
2020-05-25 23:00:44,155   INFO  eval_tag         default
2020-05-25 23:00:44,155   INFO  eval_all         False
2020-05-25 23:00:44,155   INFO  ckpt_dir         None
2020-05-25 23:00:44,155   INFO  save_to_file     True
2020-05-25 23:00:44,155   INFO  cfg.ROOT_DIR: /home/tleyden/Development/PCDet
2020-05-25 23:00:44,155   INFO  cfg.LOCAL_RANK: 0
2020-05-25 23:00:44,155   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2020-05-25 23:00:44,155   INFO
cfg.DATA_CONFIG = edict()
2020-05-25 23:00:44,155   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2020-05-25 23:00:44,155   INFO  cfg.DATA_CONFIG.DATA_DIR: data/kitti
2020-05-25 23:00:44,155   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2020-05-25 23:00:44,155   INFO
cfg.DATA_CONFIG.NUM_POINT_FEATURES = edict()
2020-05-25 23:00:44,155   INFO  cfg.DATA_CONFIG.NUM_POINT_FEATURES.total: 4
[... 183 lines cut ...]
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:84.5649, 70.7907, 65.6831
bev  AP:81.5557, 63.4048, 61.8452
3d   AP:79.1085, 61.9706, 59.1882
aos  AP:84.28, 69.08, 64.07
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:88.1536, 71.5481, 67.0920
bev  AP:84.2010, 65.0413, 61.5829
3d   AP:80.5159, 62.7871, 58.7612
aos  AP:87.82, 69.69, 65.20
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:84.5649, 70.7907, 65.6831
bev  AP:83.6275, 68.8680, 63.5987
3d   AP:83.6275, 68.8680, 63.5987
aos  AP:84.28, 69.08, 64.07
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:88.1536, 71.5481, 67.0920
bev  AP:87.1781, 69.2666, 64.7940
3d   AP:87.1781, 69.2666, 64.7940
aos  AP:87.82, 69.69, 65.20

2020-05-25 23:04:03,887   INFO  Result is save to /home/tleyden/Development/PCDet/output/pointpillar_high_thresh/default/eval/epoch_no_number/val/default
2020-05-25 23:04:03,887   INFO  ****************Evaluation done.*****************
```
<details>

closed 2020-05-26 by tleyden

LZDSJTU, 2020-06-04:
Hi, where does the kitti_object.py come from? Is it from other project?

tleyden, 2020-06-07:
@LZDSJTU yep, from here: https://github.com/kuixu/kitti_object_vis

zjx99, 2020-11-22:
Hi, @tleyden,
As far as I know, the detection results from OpenPCDet are stored in pkl files.  However, it seems the label input of this visualization repository should be txt file in the same format as KITTI label. Did you do any conversion? 
Thanks

arxidinakbar, 2020-12-09:
@zjx99  you could get txt detection results by setting --save_to_file flags, refer README.

zjx99, 2020-12-10:
@arxidinakbar , thank you!  I got it.