open-mmlab/OpenPCDet #75: Issue when use Sync BN

thangvubk, 2020-05-26:
Thanks for the nice project.
When I turn on Sync BN argument, an error occurred ``ValueError: expected at least 3D input``. It seems Sync BN conversion does not support BatchNorm1D. 
Have you ever run the network normally with Sync BN?

sshaoshuai, 2020-06-02:
Yes, you could simply modify the pytorch source file `PYTORCH_PATH/torch/nn/modules/batchnorm.py` to support BatchNorm1D as follows: 

```
class SyncBatchNorm: 
def _check_input_dim(self, input):
 423         # if input.dim() <= 2: 
 424         #    raise ValueError('expected at least 3D input (got {}D input)'
 425         #                     .format(input.dim()))
 426         if input.dim() < 2:  # modified to support input (B, C)
 427             raise ValueError('expected at least 2D input (got {}D input)'
 428                               .format(input.dim())) 
```

thangvubk, 2020-06-03:
I got it. Thanks for your feedback.

closed 2020-06-03 by thangvubk