open-mmlab/OpenPCDet #76: How to removing the ground plane?

ymxlzgy, 2020-05-27:
Hi, Shaoshuai

I am currently working on your new project PCDet. I can see your dataset preparation part in the repo containing an optional choice named road planes. Does it mean that you use some code generating road plane in kitti dataset? If so, can you tell me where I can find this code part? Thanks a lot.

Best,

Guangyao Zhai

sshaoshuai, 2020-06-02:
Ground plane files were copied from others long long ago, and I didn't run the road plane generation. 
By the way, it is not necessary to run PCDet with ground plane, and I guess it has little effect on the performance.

xmyqsh, 2020-06-02:
It depends on the accuracy of ground plane estimation.
Imagine that we have accurate HDMap with the ground plane, the point cloud size will reduce dramatically.

I'm not sure how much will the ground plane estimation affect the GT-Sampling process.

sshaoshuai, 2020-06-02:
I think a coarse plane is already enough for GT sampling, and remove all ground points may speed up the detector but decrease the performance of 3D object detection.

u1095279221321, 2020-07-16:
> Hi, Shaoshuai
> I am currently working on your new project PCDet. I can see your dataset preparation part in the repo containing an optional choice named road planes. Does it mean that you use some [...]

在avod的开源连接中可以找到plane.zip

closed 2020-07-17 by sshaoshuai