open-mmlab/OpenPCDet #77: kitti calibration bug?

bigsheep2018, 2020-05-28:
Hello PCDet team @codyreading @sshaoshuai,

Thanks for sharing this framework.

I found some interesting labels in forms of 3d bbox based on your code.
Basically, I just visualize the 'gt_box_lidar' of the kitti_infos_xxx.pkl files and found the 3d box is not closely fitting the vehicle points' surface. Some box even cannot cover the points belong to it. 

Examples in training split are shown as follows 
000014: the box is kind of too large for the car
[image]

000082: the boxes seems cannot cover the points belong to them
[image]

There are actually more in the labels.
Do I miss something in PCDet or indeed there are some errors of the labels?

codyreading, 2020-05-28:
Hi @bigsheep2018,

Looks like these are just non-perfect labels. 

In 000014, it's hard to tell if the box is too lagre for the car, as only the LiDAR points belonging the the front of the car are visible

In 000082, the points are only slightly in front of the boxes.

I visualized these samples myself, and found the same thing.

bigsheep2018, 2020-05-29:
@codyreading 
Thanks for the reply.

closed 2020-05-29 by bigsheep2018

LZDSJTU, 2020-06-04:
> @codyreading
> Thanks for the reply.

Hi, I want to ask which tool you use to visualize the detection result. Thank you.

codyreading, 2020-06-04:
Hello,

I use [this](https://github.com/kujason/scene_vis) tool which uses VTK to visualize 3D scenes