open-mmlab/OpenPCDet #78: RuntimeError: CUDA error: device-side assert triggered

wdxpython, 2020-06-02:
File "train.py", line 149, in <module>
    main()
  File "train.py", line 142, in main
    max_ckpt_save_num=args.max_ckpt_save_num
  File "/media/adas/File/wdx/dianyun/c/PCDet-master/tools/train_utils/train_utils.py", line 81, in train_model
    leave_pbar=(cur_epoch + 1 == total_epochs)
  File "/media/adas/File/wdx/dianyun/c/PCDet-master/tools/train_utils/train_utils.py", line 36, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/media/adas/File/wdx/dianyun/c/PCDet-master/pcdet/models/__init__.py", line 25, in model_func
    ret_dict, tb_dict, disp_dict = model(input_dict)
  File "/home/adas/anaconda3/envs/pytorch1.2/lib/python3.7/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/adas/File/wdx/dianyun/c/PCDet-master/pcdet/models/detectors/second_net.py", line 47, in forward
    rpn_ret_dict = self.forward_rpn(**input_dict)
  File "/media/adas/File/wdx/dianyun/c/PCDet-master/pcdet/models/detectors/second_net.py", line 34, in forward_rpn
    **{'gt_boxes': kwargs.get('gt_boxes', None)}
  File "/home/adas/anaconda3/envs/pytorch1.2/lib/python3.7/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/media/adas/File/wdx/dianyun/c/PCDet-master/pcdet/models/bbox_heads/rpn_head.py", line 324, in forward
    ret_dict['anchors'] = torch.from_numpy(self.anchor_cache['anchors']).cuda()
RuntimeError: CUDA error: device-side assert triggered

wdxpython, 2020-06-02:
spconv== 1.0
cuda 10.0
pytorch 1.1.0

sshaoshuai, 2020-06-03:
Could you run the inference with pre-trained models successfully?

wdxpython, 2020-06-03:
> Could you run the inference with pre-trained models successfully?

I can successfully run inference using pre-trained models.

sshaoshuai, 2020-06-03:
So when did it happen for this error? After you trained with several iterations or just the first time's inference? 
Remember that the error doesn't mean you have error in this line.

wdxpython, 2020-06-03:
> So when did it happen for this error? After you trained with several iterations or just the first time's inference?
> Remember that the error doesn't mean you have error in this line.

After several iterations of training it will go wrong. After I added this CUDA_LAUNCH_BLOCKING=1, the error message is
x = x.dense()
File "/home/adas/anaconda3/envs/pytorch/lib/python3.7/site-packages/spconv/init.py", line 73, in dense
res = scatter_nd(self.indices.long(), self.features, output_shape)
File "/home/adas/anaconda3/envs/pytorch/lib/python3.7/site-packages/spconv/init.py", line 41, in scatter_nd
ret[slices] = updates.view(*output_shape)
RuntimeError: CUDA error: device-side assert triggered. This error bothered me for a long time. Yesterday I changed python from 3.7 to 3.6, pyotch1.4, spconv1.0. This error disappeared, but a new error appeared. Then I executed sudo apt-get upgrade. There is no error in training until 22epoch, the training is not over, I will tell you later if I can successfully train.

sshaoshuai, 2020-06-03:
Great to hear that!

wdxpython, 2020-06-12:
> Great to hear that!

Excuse me, I want to submit my test results on Kitti's official website. What is the format of the submission?

sshaoshuai, 2020-07-17:
Please refer to here for generaing test results for submission: https://github.com/open-mmlab/OpenPCDet/issues/125

closed 2020-07-21 by sshaoshuai