open-mmlab/OpenPCDet #79: Tools to visualize the detection result.

LZDSJTU, 2020-06-04:
Hi, I want to ask which tools can visualize the detection result.

Additionally, if I train the network in server and want to visualize in the windows10, is there any tool recommended?

If anyone has recommended tools, comment in this topic. Thank you very much.

penghao1990, 2020-06-05:
kittiviewer  in  traveller59 /second.pytorch. (html+css+js+flask)     :)

ltphy, 2020-06-09:
You can use [http://www.open3d.org/docs/release/tutorial/Advanced/customized_visualization.html](url) and self implemented your own visualization with pickle file.

MartinHahner88, 2020-07-11:
@sshaoshuai: Do you still plan to release/integrate your visualization code to this repository?

sshaoshuai, 2020-07-11:
@MartinHahner88 ，ok，I will prepare a simple visualization codes to `PCDet` soon.

sshaoshuai, 2020-07-17:
Hi all, a simple visualization tool and [quick demo](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/DEMO.md) is available now.

closed 2020-07-17 by sshaoshuai

Petros626, 2025-09-13:
[image]