open-mmlab/OpenPCDet #80: get_direction_target@dir_offset

penghao1990, 2020-06-05:
Nice work!
File: /PCDet/pcdet/models/bbox_heads/rpn_head.py
```python
@staticmethod
    def get_direction_target(anchors, reg_targets, one_hot=True, dir_offset=0, num_bins=2):
        batch_size = reg_targets.shape[0]
        anchors = anchors.view(batch_size, -1, anchors.shape[-1])
        rot_gt = reg_targets[..., 6] + anchors[..., 6]
        offset_rot = common_utils.limit_period_torch(rot_gt - dir_offset, 0, 2 * np.pi)
        dir_cls_targets = torch.floor(offset_rot / (2 * np.pi / num_bins)).long()
        dir_cls_targets = torch.clamp(dir_cls_targets, min=0, max=num_bins - 1)

        if one_hot:
            dir_targets = torch.zeros(*list(dir_cls_targets.shape), num_bins, dtype=anchors.dtype,
                                      device=dir_cls_targets.device)
            dir_targets.scatter_(-1, dir_cls_targets.unsqueeze(dim=-1).long(), 1.0)
            dir_cls_targets = dir_targets
        return dir_cls_targets
```
Why dir_offset is needed here and set to pi/4.
Train model: pointpillars
Thanks.

xmyqsh, 2020-06-10:
@penghao1990 
Good question!!!
You could try to change the `dir_offset` from `pi/4` to `pi/2`, then you could get a very good VEHICLE heading classification and poor PEDESTRIAN heading classification.

Suddenly, you will find that it is straight forward!

From the perspective of statistics, the heading of VEHICLE could be clustered into two groups, heading equals `0` and heading equals `pi`, and `pi/2` and `-pi/2` for PEDESTRIAN. Though the cluster of the latter one is not as tight as the former.

And now, everyone will choose dir_offset equals `+/-pi/4`

penghao1990, 2020-06-30:
Thank you very much for your help. ^_^

closed 2020-06-30 by penghao1990

VishalBalaji321, 2023-01-23:
Hi, I still couldnt get the reason why we need pi/4 specifically.....why cant it be completely ignored? No mention of this offset has been made in the original pointpillars paper. Explanation from your side would be highly appreciated !!

Thanks in advance!!