open-mmlab/OpenPCDet #83: RuntimeError: cuda runtime error (9) : invalid configuration argument at /pytorch/aten/src/THC/generic/THCTensorMath.cu:35

MohamedRamzy1, 2020-06-10:
Hello

I face an error when I run this command to evaluate kitti dataset on a single GPU.

`python test.py --cfg_file cfgs/PartA2_fc.yaml --batch_size 4 --ckpt PartA2_fc.pth`

The error: 

`Traceback (most recent call last):
  File "test.py", line 181, in <module>
    main()
  File "test.py", line 177, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id)
  File "test.py", line 59, in eval_single_ckpt
    model, test_loader, epoch_id, logger, result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/home/tools/eval_utils/eval_utils.py", line 46, in eval_one_epoch
    pred_dicts, ret_dict = model(input_dict)
  File "/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/pcdet/models/detectors/PartA2_net.py", line 112, in forward
    batch_size, voxel_centers, coords, rpn_ret_dict, input_dict
  File "/home/pcdet/models/detectors/PartA2_net.py", line 98, in forward_rcnn
    rcnn_ret_dict = self.rcnn_net.forward(rcnn_input_dict)
  File "/home/pcdet/models/rcnn/partA2_rcnn_net.py", line 553, in forward
    pooled_part_features, pooled_rpn_features = self.roiaware_pool(rois, rcnn_dict)
  File "/home/pcdet/models/rcnn/partA2_rcnn_net.py", line 510, in roiaware_pool
    cur_roi, cur_voxel_centers, cur_rpn_features, pool_method='max'
  File "/home/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 16, in forward
    self.out_size, self.max_pts_each_voxel, pool_method)
  File "/home/pcdet/ops/roiaware_pool3d/roiaware_pool3d_utils.py", line 44, in forward
    pooled_features = pts_feature.new_zeros((num_rois, out_x, out_y, out_z, num_channels))
RuntimeError: cuda runtime error (9) : invalid configuration argument at /pytorch/aten/src/THC/generic/THCTensorMath.cu:35
`

The code is run on a container based on the docker image provided in the repo.

sshaoshuai, 2020-06-17:
I have no idea about this error. What's your PyTorch version and GCC version and CUDA version?

zwqnju, 2020-07-10:
Can you print these arguments: (num_rois, out_x, out_y, out_z, num_channels) ?

Angericky, 2020-07-17:
I met the same problem with pytorch 1.1.0, cudatoolkit 10.0 and cudnn 7.5.1. Did you solve it?

MohamedRamzy1, 2020-07-17:
I think the problem is related to the data .. There is a single corrupted image in kitti dataset the causes this error so I removed it from the dataset and then problem solved.

closed 2020-07-30 by sshaoshuai