open-mmlab/OpenPCDet #86: Error when ''pip install -r requirements.txt''

fyaft2012, 2020-06-16:
Hi @sshaoshuai , thanks for your excellent work! However, when I install packages in requirement.txt. It shows error:
------------------------------------------------------------------------------------------------------------
Requirement already satisfied: numpy in /home/lion/anaconda3/envs/pcdet/lib/python3.7/site-packages (from -r requirements.txt (line 1)) (1.18.1)
Requirement already satisfied: torch>=1.1 in /home/lion/anaconda3/envs/pcdet/lib/python3.7/site-packages (from -r requirements.txt (line 2)) (1.1.0)
ERROR: Could not find a version that satisfies the requirement spconv (from -r requirements.txt (line 3)) (from versions: none)
ERROR: No matching distribution found for spconv (from -r requirements.txt (line 3))
---------------------------------------------------------------------------------------------------------------------------

I am confused that why it says can not find a version for spconv. 
My server and Anaconda environment: Ubuntu 16.04, python 3.7, pytorch 1.1.0
Hope get help from you, thanks!

MartinHahner88, 2020-06-16:
you need to install spconv before running `pip install -r requirements.txt`

How you have to install it is explained in the readme just below the line of `pip install -r requirements.txt`.

fyaft2012, 2020-06-16:
Hi @MartinHahner88, thank you for timely answer! I will try to install spconv first. Thanks.

closed 2020-06-30 by sshaoshuai