open-mmlab/OpenPCDet #89: how to run test.py on the /testing dataset

Royaler, 2020-06-18:
@sshaoshuai

Royaler, 2020-06-18:
and how to submit the test result to kitti website

sshaoshuai, 2020-06-30:
In the latest code, you could set `DATA_SPLIT['test'] = 'test'`, and also set the `INFO_PATH['test']=[kitti_infos_test.pkl]` to generate the predictions for test split. 
To submit to kitti website, you could run `test.py` with `--save_to_file` parameter to save the required prediction results to files.

closed 2020-07-01 by sshaoshuai

sgrrlll, 2023-10-13:
> In the latest code, you could set `DATA_SPLIT['test'] = 'test'`, and also set the `INFO_PATH['test']=[kitti_infos_test.pkl]` to generate the predictions for test split. To submit to kitti website, [...]

Hello, I set --save_to_file default to true, but there is still no result in the txt file, do i have to get the test results on the kitti official website
[image]
It also generates a result.pkl file, what does this do?

Royaler, 2023-10-13:
黄佳遥已收到你的来信。