open-mmlab/OpenPCDet #91: RuntimeError: CUDA error: invalid configuration argument

Divadi, 2020-06-23:
Hello, thank you for your work.
I was running the repo on a different, larger dataset, but I continue to run into this issue somewhat randomly. For instance, this run failed 221 batches in. Do you know what may be the problem?
```
  File "train.py", line 149, in <module>                           
    main()
  File "train.py", line 142, in main
    max_ckpt_save_num=args.max_ckpt_save_num
  File "/home/david/src/projects/reproduce_methods/PCDet/tools/train_utils/train_utils.py", line 81, in train_model
    leave_pbar=(cur_epoch + 1 == total_epochs)
  File "/home/david/src/projects/reproduce_methods/PCDet/tools/train_utils/train_utils.py", line 36, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/__init__.py", line 25, in model_func
    ret_dict, tb_dict, disp_dict = model(input_dict)
  File "/home/david/anaconda3/envs/p3.6_t1.1_spconv8da/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/david/anaconda3/envs/p3.6_t1.1_spconv8da/lib/python3.6/site-packages/torch/nn/parallel/distributed.py", line 376, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/david/anaconda3/envs/p3.6_t1.1_spconv8da/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/detectors/PartA2_net.py", line 112, in forward
    batch_size, voxel_centers, coords, rpn_ret_dict, input_dict
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/detectors/PartA2_net.py", line 98, in forward_rcnn
    rcnn_ret_dict = self.rcnn_net.forward(rcnn_input_dict)
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/rcnn/partA2_rcnn_net.py", line 547, in forward
    targets_dict = self.assign_targets(batch_size, rcnn_dict)
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/rcnn/partA2_rcnn_net.py", line 27, in assign_targets
    targets_dict = proposal_target_layer(rcnn_dict, roi_sampler_cfg=self.rcnn_target_config)
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/model_utils/proposal_target_layer.py", line 14, in proposal_target_layer
    sample_rois_for_rcnn(rois, gt_boxes, roi_raw_scores, roi_labels, roi_sampler_cfg)
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/model_utils/proposal_target_layer.py", line 77, in sample_rois_for_rcnn
    iou3d = iou3d_nms_utils.boxes_iou3d_gpu(cur_roi, cur_gt[:, 0:7])  # (M, N)
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 54, in boxes_iou3d_gpu
    vol_a = (boxes_a[:, 3] * boxes_a[:, 4] * boxes_a[:, 5]).view(-1, 1)
RuntimeError: CUDA error: invalid configuration argument
```

sshaoshuai, 2020-06-24:
It means your current training scene doesn't have any GT box. In our lasted `PCDet` v0.2.0, we have modified the dataloader to avoid this situation during training (refer to [here](https://github.com/open-mmlab/OpenLidarPerceptron/blob/master/pcdet/datasets/dataset.py#L79)). You could use a similar strategy to avoid it.

Divadi, 2020-06-26:
Thank you, I'll use that method.

I had gotten rid of the scenes w/o GT boxes for further training, but I ended up getting the following issue.
This error still seems to shows with the modification in the dataloader.
```
  File "train.py", line 149, in <module>                                                                                                                                                                    
    main()                                                                                                                                                                                                  
  File "train.py", line 142, in main                                                                                                                                                                        
    max_ckpt_save_num=args.max_ckpt_save_num                                                                                                                                                                
  File "/home/david/src/projects/reproduce_methods/PCDet/tools/train_utils/train_utils.py", line 81, in train_model                                                                                         
    leave_pbar=(cur_epoch + 1 == total_epochs)                                                                                                                                                              
  File "/home/david/src/projects/reproduce_methods/PCDet/tools/train_utils/train_utils.py", line 36, in train_one_epoch                                                                                     
    loss, tb_dict, disp_dict = model_func(model, batch)                                                                                                                                                     
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/__init__.py", line 25, in model_func                                                                                                  
    ret_dict, tb_dict, disp_dict = model(input_dict)                                                                                                                                                        
  File "/home/david/anaconda3/envs/p3.6_t1.1_spconv8da/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__                                                                       
    result = self.forward(*input, **kwargs)                                                                                                                                                                 
  File "/home/david/anaconda3/envs/p3.6_t1.1_spconv8da/lib/python3.6/site-packages/torch/nn/parallel/distributed.py", line 376, in forward                                                                  
    output = self.module(*inputs[0], **kwargs[0])                                                                                                                                                           
  File "/home/david/anaconda3/envs/p3.6_t1.1_spconv8da/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__                                                                       
    result = self.forward(*input, **kwargs)                                                                                                                                                                 
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/detectors/PartA2_net.py", line 112, in forward                                                                                        
    batch_size, voxel_centers, coords, rpn_ret_dict, input_dict                                                                                                                                             
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/detectors/PartA2_net.py", line 98, in forward_rcnn                                                                                    
    rcnn_ret_dict = self.rcnn_net.forward(rcnn_input_dict)                                                                                                                                                  
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/rcnn/partA2_rcnn_net.py", line 547, in forward                                                                                        
    targets_dict = self.assign_targets(batch_size, rcnn_dict)                                                                                                                                               
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/rcnn/partA2_rcnn_net.py", line 27, in assign_targets                                                                                  
    targets_dict = proposal_target_layer(rcnn_dict, roi_sampler_cfg=self.rcnn_target_config)                                                                                                                
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/model_utils/proposal_target_layer.py", line 14, in proposal_target_layer                                                              
[... cut ...]
  File "/home/david/src/projects/reproduce_methods/PCDet/pcdet/models/model_utils/proposal_target_layer.py", line 146, in sample_rois_for_rcnn                                                              
    gt_of_bg_rois = cur_gt[gt_assignment[bg_inds]]                                                                                                                                                          
IndexError: index is out of bounds for dimension with size 0
```

Oddly, it seems like a similar issue but it seems to happen randomly once ever few epochs. Do you know why this may be the case?

sshaoshuai, 2020-06-30:
Maybe you could also print out the shape of cur_gt, and also print out the content of gt_assignment and bg_inds. I doubt it is a similar issue.

Divadi, 2020-06-30:
In v0.2, I am not longer getting this issue, so it's likely not a big concern anymore.

closed 2020-06-30 by Divadi