open-mmlab/OpenPCDet #92: How can we train the net without plane data?

lgldl, 2020-06-23:
According to the instructions, the KITTI dataset can have no planes data. However，while we train the SECOND Net without plane data, the error occured, which is as follows:

Traceback (most recent call last):
  File "/home/lgl/3D_object_detection/PCDet/tools/train.py", line 149, in <module>
    main()
  File "/home/lgl/3D_object_detection/PCDet/tools/train.py", line 142, in main
    max_ckpt_save_num=args.max_ckpt_save_num
  File "/home/lgl/3D_object_detection/PCDet/tools/train_utils/train_utils.py", line 81, in train_model
    leave_pbar=(cur_epoch + 1 == total_epochs)
  File "/home/lgl/3D_object_detection/PCDet/tools/train_utils/train_utils.py", line 18, in train_one_epoch
    batch = next(dataloader_iter)
  File "/home/lgl/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 345, in __next__
    data = self._next_data()
  File "/home/lgl/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 856, in _next_data
    return self._process_data(data)
  File "/home/lgl/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 881, in _process_data
    data.reraise()
  File "/home/lgl/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/_utils.py", line 394, in reraise
    raise self.exc_type(msg)
FileNotFoundError: Caught FileNotFoundError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/lgl/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/lgl/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/lgl/anaconda3/envs/PCDet/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/lgl/3D_object_detection/PCDet/pcdet/datasets/kitti/kitti_dataset.py", line 460, in __getitem__
    example = self.prepare_data(input_dict=input_dict, has_label='annos' in info)
  File "/home/lgl/3D_object_detection/PCDet/pcdet/datasets/dataset.py", line 96, in prepare_data
    if cfg.DATA_CONFIG.AUGMENTATION.DB_SAMPLER.USE_ROAD_PLANE else None
  File "/home/lgl/3D_object_detection/PCDet/pcdet/datasets/kitti/kitti_dataset.py", line 55, in get_road_plane
    with open(plane_file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/lgl/3D_object_detection/PCDet/data/kitti/training/planes/004467.txt'

Can you tell me how to solve it

Divadi, 2020-06-23:
You should set 
https://github.com/sshaoshuai/PCDet/blob/19cc5b801ccf3a64d46a84718fcbbc4108f5d29d/tools/cfgs/second.yaml#L47
to be False if you haven't done so

closed 2020-06-30 by sshaoshuai

Nerdmust, 2020-11-24:
> You should set
> https://github.com/sshaoshuai/PCDet/blob/19cc5b801ccf3a64d46a84718fcbbc4108f5d29d/tools/cfgs/second.yaml#L47
> to be False if you haven't done so

thanks！